PX4/PX4-Autopilot #12015: RC Loss Failsafe not triggered

tops4u, 2019-05-13:
**Describe the bug**
Today I had a RC Loss for around 10 sec (see Log). Actuall the Failsafe for RC Loss is set to 0.5sec to trigger Return to Land:

[image]

But this did not happen. Instead, 

**Expected behavior**
Even when in Mission Mode, I would assume that RC Link Lost would trigger the Failsafe Action?

**Log Files and Screenshots**
https://review.px4.io/plot_app?log=e91a4c2f-670f-4832-b6e2-ff851ac044ac

RC Lost from Log:
[image]

Mission Continued even after RC Lost:
[image]

**Drone (please complete the following information):**
Generic HEXA
CUAV Pixhack v5
PX4 1.9 rc2

dagar, 2019-05-13:
A loss of RC is tolerated if you're flying a mission that hasn't finished and you have a valid data link.

dagar, 2019-05-13:
Standby, there's something not quite right here.

dagar, 2019-05-13:
In the current logic for auto mission.

 - If data link loss is configured (NAV_DLL_ACT) and you lose your data link it will trigger return (RTGS).
 - if data link loss is not configured **AND** RC is lost **AND** the mission is finished it will trigger return (RCRECOVER)

https://github.com/PX4/Firmware/blob/f067ca0d8f77af3b60c2814e02dfd487e735ae66/src/modules/commander/state_machine_helper.cpp#L500-L517

I can see the dependence on mission finished potentially being unexpected. Thoughts?

hamishwillee, 2019-05-13:
>

bresch, 2019-05-20:
@hamishwillee Yes, those triggers are mode sensitive; I realized that the datalink loss action is only triggered when in mission (which makes sense, but is - AFAIK - not documented).

tops4u, 2019-05-20:
> A loss of RC is tolerated if you're flying a mission that hasn't finished and you have a valid data link.

@dagar how could I check if there was a valid Datalink? I doubt that there was since my current link is a CUAV WLAN Board which connects to my iPad. I doubt that the WiFi Connection was working while the Futaba FASTest Link failed in a bit more than 450m distance from home.

julianoes, 2019-08-19:
@tops4u do you have some component like a companion computer on your vehicle that would regularly send heartbeats to the autopilot (even when the connection to the ground is lost)?

julianoes, 2019-08-19:
@tops4u another observation is that the data link was lost 3 times so presumably it came back in-between as well. Unfortunately, the moment it comes back is not logged because it's only a info log output and not a warning.

tops4u, 2019-08-19:
@julianoes can't say as the iPad was in the Trunk of my car... So i guess the 450m was anyhow a "stretch" connection for WLAN ;-) 
So maybe we just drop this? As long as the Aircraft will RTH when both links are lost, it's ok for me. Preferably this could be configured? As in my case I just use the iPad to plan the mission and check for Arming Problem, then I just use the Remote. So my WLAN connection was more a kind of accident. Best case would be to configure if missions continue when RC is lost - currently I was not aware that this is the case as it is actually shown in QGC.

julianoes, 2019-08-19:
Yes we can drop it but I want to test (and add a test) to check that sort of behaviour at some point.

closed 2019-08-21 by julianoes