PX4/PX4-Autopilot #12016: the formula "qd_red *= q" in AttitudeControl.cpp don't match the ETH article formula

sanabj, 2019-05-14:
When reading the article "https://www.research-collection.ethz.ch/bitstream/handle/20.500.11850/154099/eth-7387-01.pdf" on page 11, the formula number 46 is "qcmd,red = q · qe,red." but in the file "AttitudeControl.cpp" it is used in line 75 as "qd_red *= q" that means  "qcmd,red = qe,red.q"
is there any explication particularely if we know that quaternion multiplication is not commutative

julianoes, 2019-08-12:
FYI @bresch

bresch, 2020-02-09:
@MaEtUgR Could you please check that?

Pedro-Roque, 2020-05-02:
Came across this now, I will take a look too @bresch @MaEtUgR

Pedro-Roque, 2020-05-02:
So, this seems to be correct, however the multiplication order in the code, and subsequent Quatf seems misleading. On the matrix/Quaternion.hpp, line 255:
    ```
Quaternion operator*(const Quaternion &q) const
    {
        const Quaternion &p = *this;
        return {
            p(0) * q(0) - p(1) * q(1) - p(2) * q(2) - p(3) * q(3),
            p(0) * q(1) + p(1) * q(0) + p(2) * q(3) - p(3) * q(2),
            p(0) * q(2) - p(1) * q(3) + p(2) * q(0) + p(3) * q(1),
            p(0) * q(3) + p(1) * q(2) - p(2) * q(1) + p(3) * q(0)};
    }
```
@jgoppert , can you take a look here? your format seems to be 
![image](https://user-images.githubusercontent.com/5076811/80872129-65e72000-8cb0-11ea-8f3c-2859e9b3b0a8.png)
but there seems to exist a transposition on the vector part of Q, that is Q(2:4,2:4) seem to be transposed w.r.t. Trawny's method, here http://mars.cs.umn.edu/tr/reports/Trawny05b.pdf .

Does this imply permutation of the quaternion multiplication? If not, then the multiplication being made is **q*p** , which would be correct.

Pedro-Roque, 2020-05-02:
It seems to me that it is correct as is, as it depends on the frame wrt which the quaternions are defined. Changing to the suggested solution - exactly as the paper, `qd_red = q*qd_red;` - induces an oscillatory behavior (tested in SITL) that suggest that there is a transformation problem. @bresch @MaEtUgR can you double check this?

I'd leave as is and close the issue.

Pedro-Roque, 2020-05-08:
Issued close. @bresch @MaEtUgR let me know if you disagree with this.

closed 2020-05-08 by Pedro-Roque