PX4/PX4-Autopilot #12017: mpu9250 using accel in _accel_int.put

yukun2016, 2019-05-14:
why the input param  of the function _accel_int.put in mpu9250.cpp using the origin accel value rather than the accel value after lowpassfilter.
bool accel_notify = _accel_int.put(arb.timestamp, aval, aval_integrated, arb.integral_dt);

        arb.x = _accel_filter_x.apply(x_in_new);
	arb.y = _accel_filter_y.apply(y_in_new);
	arb.z = _accel_filter_z.apply(z_in_new);

	matrix::Vector3f aval(**x_in_new, y_in_new, z_in_new**);
	matrix::Vector3f aval_integrated;

	bool accel_notify = _accel_int.put(arb.timestamp, aval, aval_integrated, arb.integral_dt);

or it should be:
      
        arb.x = _accel_filter_x.apply(x_in_new);
	arb.y = _accel_filter_y.apply(y_in_new);
	arb.z = _accel_filter_z.apply(z_in_new);

	matrix::Vector3f aval(**arb.x, arb.y, arb.z**);
	matrix::Vector3f aval_integrated;

	bool accel_notify = _accel_int.put(arb.timestamp, aval, aval_integrated, arb.integral_dt);

julianoes, 2019-05-14:
Good question, thanks for the issue.

@dagar does that make sense to you?

dagar, 2019-05-14:
There are separate paths for the estimator (integrated data) and the controllers (low pass filtered). This will hopefully be clearer in the future once split into separate messages so that we can run the rate controller much faster.

julianoes, 2019-05-14:
Oh right, and the filtered values are put into the ringbuffer here, right?
https://github.com/PX4/Firmware/blob/01fdd00c41dd5986e8b1faf0f735b406aed629b4/src/drivers/imu/mpu9250/mpu9250.cpp#L1236

dagar, 2019-05-14:
Yes, but the ringbuffers (and reading data from /dev/accelX) aren't even used anymore. The only data PX4 actually uses is coming out through orb where both the integrated and filtered data are in the same message. I'm hoping we can clean this up soon.

julianoes, 2019-05-14:
Ok, so is the right (filtered) data used for uORB?

dagar, 2019-05-14:
Both are used in uORB (but should be split). 

``` C
struct sensor_accel_s {
	uint64_t timestamp;
	uint64_t error_count;
	uint32_t device_id;

	float x;                 // low pass filtered data used by controllers
	float y;
	float z;

	uint32_t integral_dt;
	float x_integral;        // integrated data for the estimator, not low pass filtered
	float y_integral;
	float z_integral;

	float temperature;

	float scaling;
	int16_t x_raw;
	int16_t y_raw;
	int16_t z_raw;
};
```

yukun2016, 2019-05-14:
Thanks for your reply
Servals days ago,I record the accel and gyro value by set sd record mode.
After fly some times,I found the vibe[2] is around 0.2.
I found the ekf using accelerometer_m_s2  rather sensor value in ekf2_main.cpp ,But the log show the accel of arb.x arb.y and arb.z is less more than accelerometer_m_s2[0],accelerometer_m_s2[1] and accelerometer_m_s2[2].
So I found the code blow in voted_sensor_update.cpp .the value accelerometer_m_s2 is using x_integral rather than accel_report x.
`
if (accel_report.integral_dt != 0) {

				// convert the delta velocities to an equivalent acceleration before application of corrections
				float dt_inv = 1.e6f / accel_report.integral_dt;
				accel_data = matrix::Vector3f(accel_report.x_integral * dt_inv,
							      accel_report.y_integral * dt_inv,
							      accel_report.z_integral * dt_inv);

				_last_sensor_data[uorb_index].accelerometer_integral_dt = accel_report.integral_dt;

			} else {
				// using the value instead of the integral (the integral is the prefered choice)

				// Correct each sensor for temperature effects
				// Filtering and/or downsampling of temperature should be performed in the driver layer
				accel_data = matrix::Vector3f(accel_report.x, accel_report.y, accel_report.z);

				// handle the cse where this is our first output
				if (_last_accel_timestamp[uorb_index] == 0) {
					_last_accel_timestamp[uorb_index] = accel_report.timestamp - 1000;
				}

				// approximate the  delta time using the difference in accel data time stamps
				_last_sensor_data[uorb_index].accelerometer_integral_dt =
					(accel_report.timestamp - _last_accel_timestamp[uorb_index]);
			}

			// handle temperature compensation
			if (!_hil_enabled) {
				if (_temperature_compensation.apply_corrections_accel(uorb_index, accel_data, accel_report.temperature,
						offsets[uorb_index], scales[uorb_index]) == 2) {
					_corrections_changed = true;
				}
			}

			// rotate corrected measurements from sensor to body frame
			accel_data = _board_rotation * accel_data;

			_last_sensor_data[uorb_index].accelerometer_m_s2[0] = accel_data(0);
			_last_sensor_data[uorb_index].accelerometer_m_s2[1] = accel_data(1);
			_last_sensor_data[uorb_index].accelerometer_m_s2[2] = accel_data(2);
`
Then I found the code in mpu9250.cpp,and I tried to changed them like the first issue i wrote .After fly some time ,the vibe[2] become 0.02.
So i'm confused about this.

closed 2019-08-26 by stale[bot]