PX4/PX4-Autopilot #12024: Crash while building SITL

simounours, 2019-05-15:
Hello,
I installed the development environment with gazebo with the script on the wiki and once I tried to compile with

 "make px4_sitl_default gazebo''

It stops at step 3 and Ubuntu become totally irreaponsive,  I have to hard reboot. 

It's Ubuntu on a clean installation. 

Is that a known issue ? 

Thanks.

mhkabir, 2019-05-15:
I ran into this as well on a clean install - seems like something in the sitl_gazebo build needs an insane amount of RAM.

bys1123, 2019-05-15:
It needs 2GB RAM/core. Means if you build this at make -j4, it will need 8GB RAM.

simounours, 2019-05-15:
it is really confusing ! I used it on the same computer in a virutal machine in windows and switched to a clean install of ubuntu due to performances problems in gazebo.

So only hope is tomake -j2 or increase the swap to help my 8gb of RAM ?

closed 2019-05-15 by LorenzMeier