PX4/PX4-Autopilot #12027: Running PX4 SITL simulation on NVIDIA DGX

samarth8927, 2019-05-16:
I am trying to run PX4 SITL and MAVROS on NVIDIA DGX. Help needed.

julianoes, 2019-05-16:
You need to describe what you tried, what worked, what didn't work and what your question is.

samarth8927, 2019-05-16:
I have actually successfully worked with PX4 SITL gazebo and MAVROS on normal PC. Now i want to with the same on NVIDIA DGX-1. I tried finding sources for this on the internet but was unable to get any. If any sources or links can be provided that would be helpful.

julianoes, 2019-05-16:
I think you need to follow instructions for the NVIDIA DGX-1. I think this is not really a PX4 issue unless something with PX4 is wrong which prevents it from running on that platform. This is more an open question that's probably more appropriate to ask in https://discuss.px4.io or on the PX4 slack.

closed 2019-05-16 by julianoes