PX4/PX4-Autopilot #12030: Thrust Spikes

mikehpark927, 2019-05-16:
**Describe the bug**
I have a vehicle that has a pixracer and an Intel up core board connected together via that UART port (baud rate 115200). I am publishing to mavros/local_position/pose a set-point position. It starts out fine, but then after a couple of seconds it has thrust spikes. I am running mavros on the up core board and I am getting a RTT time sync warning, but I am not sure if that is the cause of this issue (wondering if the RTT time sync warning is causing a lag in sending the set points). 

**Log Files and Screenshots**
[image]

[image]

bkueng, 2019-05-17:
Can you provide the full log?
I suggest a higher baudrate, like 921600.

mikehpark927, 2019-05-17:
Here is the link to the full log:
https://review.px4.io/plot_app?log=216aa349-50fd-4d60-8466-9ebf5fb151a8

The UART port only supports up to 115200.

bkueng, 2019-05-21:
The problem is that you have in-air ground_contact detected, causing the thrust to drop and then spike. You need to tune the hover throttle (`MPC_THR_HOVER`). Looks like about 33% from the log.

Also from the log I see mag inteferences that cause problems with heading estimation. Are you using an external mag?

mikehpark927, 2019-05-21:
Thank you very much! Lowering the hove thrust fixed the issue. We are currently using only the internal mag, but we will be using an external mag for future tests.

closed 2019-05-21 by mhkabir