PX4/PX4-Autopilot #12033: Multiple battery module support

ryanjAA, 2019-05-17:
It would be useful for anyone two different battery or battery types onboard to monitor those batteries, current, etc. think or a main battery and a backup or payload battery of a differing cell count, etc. 

Various Pixhawk hardware (Pixhawk 4, Cube, Pixhack v5) have multiple power ports but they, from what I gather are used for redundancy, not different monitoring solutions under px4. 

We should be able to monitor power port 1 and power port 2 battery input and current and the corresponding battery_status message so qgc can display it but we will also need parameters to configure both power sources and eventually fail-safe logic tied to one or both. 

According to this, https://github.com/PX4/Firmware/issues/8578#issuecomment-355316273 from @dagar the codebase can already handle 4x power bricks but there are no parameters for them. We should expose them so qgc will report them so at a minimum that would work quite quickly. 

As qgc is currently stands, it looks for BATTERY_STATUS.id=X if x is 0 it reports the first battery, if x is >0 it displays the second battery. This was done to coincide with how Ardupilot reports multiple batteries so essentially we just need to publish a battery status message with an id greater than 0 and parameters to set the amps per volt and voltage divider for the second port and in theory it should work. Failsafe and related can come later. 

A qgc front end would be great but as we are seemingly missing something on the firmware side, that probably comes first.

ryanjAA, 2019-05-17:
I should add that I tried testing what happens on fmuv5 with two bricks. It always reads from port 1 if two are plugged in. If you remove one it reports the other one instantly but not two at a time.

hamishwillee, 2019-08-15:
@ryanjAA This still being worked on?

julianoes, 2019-08-16:
FYI @ItsTimmy

ItsTimmy, 2019-08-19:
Some functionality was added in #12034, and the calibration parameters are being added in #12551.

closed 2020-01-22 by LorenzMeier

brunweil, 2023-01-15:
When two Power-modules are connected, the two voltage are readable but only the first "active" one show current...
I work with two concomitant alimentations (Solar and Li-ion) Substract the second to the first would well inform the system of the battery state.

dagar, 2023-01-15:
@brunweil if you have a particular feature request (or even possible bug?) could you please open a new issue and share more details like your particular board, maybe a flight log to see the data, etc. Typically the dual modules have been used for redundant general batteries, but more specific context per module should also be possible. We might also have to keep in mind the underlying power selector hardware in a lot of these boards, but that's a detail we can get into later.

brunweil, 2023-01-15:
> @brunweil if you have a particular feature request (or even possible bug?) could you please open a new issue and share more details like your particular board, maybe a flight log to see the data, [...]

Thank you very Much Dagar !
I can try in this way....
But simultaneous use of sevral baterry can make sens in many case... Simultaneous monitoring too of course.