PX4/PX4-Autopilot #12035: FW review RTL loiter behaviour

dagar, 2019-05-17:
From https://discuss.px4.io/t/plane-flyaway-recovery-rtl-loiter-then-flew-away/10991.

https://logs.px4.io/plot_app?log=a204d10b-a40d-4c05-9bb6-6f2c23bf574c

> I found that putting a FIXED WING LANDING PATTERN in the mission FROM the preset “pattern” icon/button, then setting the RTL mode to head for the landing 1st (NOT head directly to home), causes the plane to go to that loiter-to-land pattern and subsequently line up to the runway and land correctly. That preset FW landing pattern is very nice - much nicer than basic waypoint and land. thx to the coder!
> .
> However, even with this fw landing pattern in place and working as such, if i tried to use the RTL mode that just flies directly to home, with the RTL conditions set to loiter for (time) 45sec then land, the plane would loiter for 45secs, then proceed to fly away in the direction of its choice. BUG!
> .

dagar, 2019-05-17:
TODO: review FW RTL behavior, and consider requiring a planned landing by default if RTL can land at all.

1Gump, 2019-05-17:
Before I added the "planned landing" (from the "pattern" icon/menu), I had a "land" waypoint with a preceding waypoint that would line the plane up for the runway. This worked sufficiently to land the plane in a mission, but would not work with the RTL cmd.

1Gump, 2019-06-13:
@dagar , did 413acc5 fix this?