PX4/PX4-Autopilot #12036: Battery Status is not recognized by px4 (Battery is n/a)

hsu-ret, 2019-05-17:
**What have I done:**
The last two day I worked with a pixhawk 2.1 aka the Cube to enable indoor flight in our Vicon test area with a ROS connection. Everything was fine I also tracked the battery state - measured values and calibrated the system. Yesterday when I wanted to try a test flight I exchanged the battery from an 3S to an 4S LiPo and adjusted the parameters in QGC. I placed the drone in the test area booted the system checked cell voltage and wanted to adjust the Volt-Divider when I realized I had no current measurement of the battery (Voltage and Consumption were shown as n/a). So I typed in all parameters again - rebooted - but still the same problem.
Afterwards I changed back to the 3S battery I used before (now 'fresh' loaded) and adjusted the parameters back to the 3S battery. But still no values are measured. The Pixhawk gains its power but doesn't know about its current state - so no takeoff possible. 
I even tried the power 2 port and reverted all parameters I could reset about the battery.

**What I checked:** 
According to my ROS message /mavros/battery it detects a Voltage of 65V. Which is definitive not correct. Its a full charged 3s battery so it should be something like 12.6V.

> header: 
>   seq: 1382
>   stamp: 
>     secs: 1558022308
>     nsecs: 979666403
>   frame_id: ''
> voltage: 65.5350036621
> current: 0.00999999977648
> charge: nan
> capacity: nan
> design_capacity: nan
> percentage: -0.00999999977648
> power_supply_status: 2
> power_supply_health: 0
> power_supply_technology: 3
> present: True
> cell_voltage: []
> location: "id0"
> serial_number: ''

**What might be the problem/ How to avoid this happening again?**
This issue seems to be rare -  at least I found nearly no topics about this (via google). Maybe someone of you know more? What I have been found is an issue where it was caused by wrong parameters but I checked them and set them to default values (which sadly didn't help).

Is it maybe some kind of broken connection? At least I test the cable connector between power port and battery system and everything was fine - maybe is something inside the Pix broken? I don't know is there any possibility to check whether all ports still behave fine?

Sadly I can just access the drone on monday again but I want to reinstall QGC and flash a new Pixhawk firmware on the drone. I don't know whether I might help but maybe some of the callibration parameters are wrong. And about the firmware version ... its the master branch for connection to an companion computer from around 4 weeks ago - don't know which firmware version it is sadly but maybe it helps...

Every help is appreciated and if this is the wrong place to seek for help - please tell me. 
I'm still fairly new to working with pixhawk/px4.

Thanks in advance and have a good weekend.

hsu-ret, 2019-05-20:
Ok it seems the issue about battery state is n/a is mostly about bad parameters in the battery settings.
Uploading a new firmware didn't helped much - but at least I have now the current master branch running.

I still don't know what was the exact solution but **I guess the voltage divider was not correct** - the default value of 10 might be a bit too far off. I set it to around 4 and its currently working fine.
**The battery is now recognized and the current voltage is shown in qgc and via mavros**.

**But there is still a problem why I don't want to close this issue - its about arming and taking off... More about not being able to arm the drone because of the battery...** 

Upon system start I receive the message: 
`Critical: Dangerously low battery! Shutting system down!`
`Critical: System does not support shutdown` 

Upon arming the drone I get the message:
`Critical: ARMING DENIED: CHECK BATTERY`

My battery state should be around 70% with 3.96V with my LiPo S3 but the current battery % is just shown as 0% (might this be the error?).
Here are my parameters of the battery settings...

[image]

If someone can see a mistake and help me it would be great.

___________________________________________________
Else I would have to comment out the safety function to test our drone and I would like to not do this.
I guess the code would be in firmware/src/modules/commander/state_machine_helper.cpp (lines921-928)

> // main battery level
		if (!status_flags.condition_battery_healthy) {
			if (prearm_ok && reportFailures) {
				mavlink_log_critical(mavlink_log_pub, "ARMING DENIED: CHECK BATTERY");
			}
			prearm_ok = false;
		}

hsu-ret, 2019-05-23:
I realized my stupid mistake ... I calibrated the voltage with the cell voltage and not with the whole system... So my _Voltage Divider_ was wrong. Now it's working xD

closed 2019-05-23 by hsu-ret

Runepx4, 2020-11-25:
Just as an FYI: At our university we got the same error message "Dangerously low battery! Shutting system down" on two drones at the same time. We could not find any issues with our battery settings. it turned out that we had 2 wireless telemetry modules on two drones that were interacting with each other. As soon as we unplugged one of them, everything was fine.