PX4/PX4-Autopilot #12037: Add Airframe configuration for NXP HoverGames KIT-HGDRONEK66

igalloway, 2019-05-17:
Would like an airframe configuration set for the NXP HoverGames KIT-HGDRONEK66

**Describe your preferred solution**
Label under quadcopter frames for "NXP-HoverGames KIT-HGDRONEK66"

**Describe possible alternatives**
Alternatively an LJI-X4 frame. This is more generic, we will likely want "forgiving" tuning on the PID settings.

**Additional context**
HoverGames will start next month, and  it should be obvious for people to select the correct frame and settings.

hamishwillee, 2019-11-15:
@igalloway Presumably you needed to generate a configuration to get it to fly? Can you PR it in? I'm not sure that anything marked "enhancement" will get much investment.

hamishwillee, 2020-02-13:
@igalloway ^^^^