PX4/PX4-Autopilot #12038: Unsupported FCU SITL

mrsiekierka, 2019-05-17:
Hello,

I had this setup running yesterday but after returning to it I am facing the following issues:

`make px4_stil_default gazebo`
and then:
`roslaunch px4 mavros_posix_sitl.launch`

In launch file:
`    <arg name="fcu_url" default="udp://:14540@localhost:14557"/>`

.bashrc:
`source ~/catkin_ws/devel_isolated/setup.bash ~/catkin_ws/src/Firmware/Tools/setup_gazebo.bash ~/catkin_ws/src/Firmware ~/catkin_ws/src/Firmware/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/catkin_ws/src/Firmware
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/catkin_ws/src/Firmware/Tools/sitl_gazebo`

Topic /mavros/status tells me that the FCU is not connected.

```
header: 
  seq: 0
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: ''
connected: False
armed: False
guided: False
mode: ''
system_status: 0

```

MODE: Unsupported FCU comes up when I try to set an offboard mode. 

Output: 
`roslaunch px4 mavros_posix_sitl.launch`

`SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
[... 228 lines cut ...]
[ INFO] [1558125552.818264992]: Plugin vision_speed_estimate loaded
[ INFO] [1558125552.821916591]: Plugin vision_speed_estimate initialized
[ INFO] [1558125552.822065624]: Plugin waypoint loaded
[ INFO] [1558125552.826377106]: Plugin waypoint initialized
[ INFO] [1558125552.826410610]: Plugin wheel_odometry blacklisted
[ INFO] [1558125552.826555999]: Plugin wind_estimation loaded
[ INFO] [1558125552.827418617]: Plugin wind_estimation initialized
[ INFO] [1558125552.827458861]: Autostarting mavlink via USB on PX4
[ INFO] [1558125552.827624724]: Built-in SIMD instructions: SSE, SSE2
[ INFO] [1558125552.827640390]: Built-in MAVLink package version: 2019.4.4
[ INFO] [1558125552.827684914]: Known MAVLink dialects: common ardupilotmega ASLUAV autoquad icarous matrixpilot paparazzi slugs standard uAvionix ualberta
[ INFO] [1558125552.827701327]: MAVROS started. MY ID 1.240, TARGET ID 1.1
SpawnModel script started
[INFO] [1558125553.142383, 0.000000]: Loading model XML from file
[INFO] [1558125553.142633, 0.000000]: Waiting for service /gazebo/spawn_sdf_model
[ INFO] [1558125554.302872583, 0.024000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1558125554.331528804, 0.052000000]: Physics dynamic reconfigure ready.
[INFO] [1558125554.349635, 0.000000]: Calling service /gazebo/spawn_sdf_model
[INFO] [1558125554.492483, 0.168000]: Spawn status: SpawnModel: Successfully spawned entity
[ INFO] [1558125554.585788539, 0.260000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1558125554.613007172, 0.288000000]: Physics dynamic reconfigure ready.
[vehicle_spawn_pawel_vyper_16100_3946803529953526077-5] process has finished cleanly
log file: /home/pawel/.ros/log/d3a0baf2-78e3-11e9-a9da-34e12ddc2ac6/vehicle_spawn_pawel_vyper_16100_3946803529953526077-5*.log
[ERROR] [1558125576.253675999, 21.916000000]: MODE: Unsupported FCU
`

mrivi, 2019-05-18:
Hi, everything you have done seems correct. Could you please check that all the paths you have set are correct? 
Try starting `posix_sitl.launch` and mavros `px4.launch` separately and see if you get any further.

mrsiekierka, 2019-05-18:
@mrivi Running `roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557` yields the same results.

All the paths in .bashrc seems to be correct, as here: https://dev.px4.io/en/simulation/ros_interface.html

```
SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:14540@127....
 * /mavros/gcs_url: 
 * /mavros/global_position/child_frame_id: base_link
 * /mavros/global_position/frame_id: map
 * /mavros/global_position/gps_uere: 1.0
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: base_link
 * /mavros/global_position/tf/frame_id: map
 * /mavros/global_position/tf/global_frame_id: earth
 * /mavros/global_position/tf/send: False
 * /mavros/global_position/use_relative_alt: True
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: base_link
[... 154 lines cut ...]
[ INFO] [1558191592.278140855, 5.968000000]: TM: Timesync mode: MAVLINK
[ INFO] [1558191592.279736223, 5.972000000]: Plugin sys_time initialized
[ INFO] [1558191592.279925191, 5.972000000]: Plugin trajectory loaded
[ INFO] [1558191592.290891387, 5.980000000]: Plugin trajectory initialized
[ INFO] [1558191592.291093422, 5.980000000]: Plugin vfr_hud loaded
[ INFO] [1558191592.292605365, 5.984000000]: Plugin vfr_hud initialized
[ INFO] [1558191592.292645010, 5.984000000]: Plugin vibration blacklisted
[ INFO] [1558191592.292802140, 5.984000000]: Plugin vision_pose_estimate loaded
[ INFO] [1558191592.305681919, 5.996000000]: Plugin vision_pose_estimate initialized
[ INFO] [1558191592.305995800, 5.996000000]: Plugin vision_speed_estimate loaded
[ INFO] [1558191592.312048972, 6.004000000]: Plugin vision_speed_estimate initialized
[ INFO] [1558191592.312275235, 6.004000000]: Plugin waypoint loaded
[ INFO] [1558191592.318352594, 6.008000000]: Plugin waypoint initialized
[ INFO] [1558191592.318426620, 6.008000000]: Plugin wheel_odometry blacklisted
[ INFO] [1558191592.318546177, 6.008000000]: Plugin wind_estimation loaded
[ INFO] [1558191592.319410261, 6.008000000]: Plugin wind_estimation initialized
[ INFO] [1558191592.319456536, 6.012000000]: Autostarting mavlink via USB on PX4
[ INFO] [1558191592.319563289, 6.012000000]: Built-in SIMD instructions: SSE, SSE2
[ INFO] [1558191592.319581980, 6.012000000]: Built-in MAVLink package version: 2019.4.4
[ INFO] [1558191592.319611935, 6.012000000]: Known MAVLink dialects: common ardupilotmega ASLUAV autoquad icarous matrixpilot paparazzi slugs standard uAvionix ualberta
[ INFO] [1558191592.319647330, 6.012000000]: MAVROS started. MY ID 1.240, TARGET ID 1.1
[ERROR] [1558191605.602404290, 19.288000000]: MODE: Unsupported FCU
[ERROR] [1558191610.617787998, 24.300000000]: MODE: Unsupported FCU

```

mrivi, 2019-05-18:
Try not to specify the fcu_url when you launch mavros. The default in the launch script works fine.

mrsiekierka, 2019-05-18:
But then the fcu_url is: FCU URL: /dev/ttyACM0:57600

Jaeyoung-Lim, 2019-05-20:
@mrsiekierka Your `Firmware` directory should not be in your `catkin_ws`. If you followed the instructions, it should be in `~/src/Firmware`

mrivi, 2019-05-20:
@mrsiekierka I was referring to `mavros_posix_sitl.launch`. 
I still feel like the system variables are not set properly. Can you copy here GAZEBO_MODEL_PATH, GAZEBO_RESOURCE_PATH, GAZEBO_PLUGIN_PATH, LD_LIBRARY_PATH and ROS_PACKAGE_PATH.

mrsiekierka, 2019-05-21:
@mrivi @Jaeyoung-Lim Thank you for helping out with this issue. I actually had sources set it up in .bashrc script what didn't work. Even though when sources where points at the right directories etc... So @mrivi suggestion was right!

In terms of having it in catkin_ws - that's right, I moved it after I had previous issues, shouldn't have done that probably. 

That's a script .sh which launches everything correctly (instead of .bashrc).

```
cd src/Firmware 
source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/sitl_gazebo

roslaunch px4 mavros_posix_sitl.launch
```

Thanks again!

closed 2019-05-21 by mrsiekierka

HenryHuYu, 2020-06-26:
> @mrsiekierka Your `Firmware` directory should not be in your `catkin_ws`. If you followed the instructions correctly, it should be in `~/src/Firmware`

I met this problem when I use Jetson Xavier do the Mavros Controller simulation part. Still working on it. I run 

make px4_sitl_default gazebo 

and 

roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557"

in two seperate terminal and there seem to be no problem.
But when I run 

roslaunch geometric_controller sitl_trajectory_track_circle.launch

I get the Unsupported FCU error.

BTW, could you explain that why the Firmware should be in a catkin_ws?