PX4/PX4-Autopilot #12051: PID in attitude - multicopter

alexcherpi, 2019-05-21:
For multicopter there is now a P gain for the attitude controller (and PID for Rate controller). Would it be possible to have PID as well for the attitude controller?

bresch, 2019-05-24:
It is possible but normally not required: when the rate controller is well tuned, you can achieve really good attitude tracking with tat simple P controller. Especially for the integral that is already implicitly there, I don't see why you would need a second one.

What problem are you facing that you want to solve with a PID attitude controller? Please provide a log.

alexcherpi, 2019-06-16:
Hi bresch, 
Indeed you're right the integral is there through the rate control Integral term.
Our drone is now flying, our bug was hardware (very vicious).
Many thanks!

closed 2019-06-16 by mhkabir