PX4/PX4-Autopilot #12054: About the position control of the offboard using mavros

kangjingjie, 2019-05-22:
Hello everyone!

I recently sent a pose to /mavros/mocap/pose via mavros.

Question 1. The /mavros/mocap/pose theme is also displayed in the rqt_graph diagram. Subscribe to /mavros/local_position/pose and find that its xyz is always around 0 and is not the same as /mavros/mocap/pose? This is why, I used the lpe firmware of px4. Is it appropriate to fuse the data?

Question 2. I did an experiment and found that the speed of the drone is too large after starting the set_position. How should I limit the speed at which the drone starts and the speed at which it reaches a certain point. If it is a parameter, what is the specific parameter?

Question 3. What is the conversion of mocap's reference system under mavros? The coordinates of the drone under mocap are (0,0,2), and the starting coordinates of the drone are (0,0,0)? If so how should I convert two coordinate systems?

Thanks!

roseyangyu, 2019-05-27:
I can set position mode if vehicle is on the ground.But when vehicle is flying,mode become altitude mode.Can you help me?

julianoes, 2019-05-27:
@roseyangyu please open a separate issue for your question and supply some details.

TobiMiller, 2019-05-29:
Question 2: The Positon Controller is a P Controller. You can set the Max horizontal velocity in xy axes. but you cannot control the specific behaviour. 
it is also possible to feed the velocity at the same time. it will be used as feedforward.

closed 2019-09-10 by stale[bot]