PX4/PX4-Autopilot #12055: Bash on Ubuntu on Windows upload the build firmware broken

huangwen0907, 2019-05-22:
I follow the [matlab/psp Setup and Configuration](https://ww2.mathworks.cn/help/supportpkg/px4/setup-and-configuration.html?s_tid=CRUX_lftnav
)

[image]

 and install the build environment. Now it can build sucessfully but can not upload the build .px4 into my hardware pixhawk-v2 (I can upload it through my 16.04 ubuntu system, so the hardware is ok ).

**FYI:** 
> I download the Firmware [mathworks/PX4-Firmware](https://github.com/mathworks/PX4-Firmware)  and use the `arm-none-eabi-gcc 5.4.1 201609019` version. 
In the windows system, qgroundcontrol  can connect my pixhawk, so the serial com is ok.

huangwen0907, 2019-05-23:
@MaEtUgR  Hi, Matthias, I am appreciate that you can give  me some suggestions here as I know you are working in the windows system. 

Now I want to upload the build firmware through the windows's bash.

hamishwillee, 2019-05-23:
@huangwen0907 

Bash is not "officially" supported, but what we know about using it is here: https://dev.px4.io/en/setup/dev_env_windows_bash_on_win.html#build-firmware

As stated:

> You can flash the custom firmware on Windows using QGroundControl or Mission Planner (it is not possible to directly flash the firmware from within the bash shell using the upload command).

In this case "not possible" means that I couldn't get it to work. As this is not "supported" probably no one else will have the time to work out whether it can be done, and if so how.

hamishwillee, 2019-05-23:
I am therefore closing this as a "known issue".

closed 2019-05-23 by hamishwillee