PX4/PX4-Autopilot #12056: VTOL Gazebo Simulation: weird MC behaviour

ThomasRigi, 2019-05-22:
Hi,

On the current (21 May 2019) master branch, I noticed that the standard VTOL on Gazebo wasn't behaving nicely when in MC mode. It calculates oscillating setpoints for pitch and roll, see log here: [https://logs.px4.io/plot_app?log=c0a081c3-c38a-4464-b584-e591baddc3cb]

This log was created after a clean build (make clean followed by make px4_sitl_default gazebo_standard_vtol) with Gazebo v9.8, QGC v3.5.2, Ubuntu 18.04.

The mission: 
[Dummy_MC_loop.zip](https://github.com/PX4/Firmware/files/3206726/Dummy_MC_loop.zip)

Any ideas where this behaviour comes from?

julianoes, 2019-05-27:
This might be related to the model changes that I had done against the vibrations on ground:
https://github.com/PX4/sitl_gazebo/pull/314

The model parameters might still not make enough sense and need improving. Or maybe it just requires some controls tuning.

Jaeyoung-Lim, 2021-03-14:
@ThomasRigi Is this still the case?

ThomasRigi, 2021-03-15:
I had completely forgotten about this stale issue. No, simulator is working fine again.

closed 2021-03-15 by ThomasRigi