PX4/PX4-Autopilot #12060: since some days ago,jmavsim need to calibration sensor.

roseyangyu, 2019-05-24:
Since some days ago,jmavsim need to calibration sensor.Two PC is both to need to calibration sensor.How do I solve that ? Oris a bug?

julianoes, 2019-05-27:
Please describe the steps to reproduce this problem. And what is the error message, and which error needs calibration?

roseyangyu, 2019-05-29:
[image]
as this and can't arm
[image]

roseyangyu, 2019-05-29:
> [image]
> as this and can't arm
> [image]

FW is 1.9.0 and 1.7

julianoes, 2019-08-27:
@roseyangyu is this still an issue?

closed 2019-08-28 by julianoes

roseyangyu, 2019-08-28:
yes,I don't solve that in old code.I can run jmavsim normal just by re-download source code.