PX4/PX4-Autopilot #12082: PREFLIGHT FAIL: MAG 2 SELFTEST FAILED

codabat, 2019-05-26:
Hi,
I am facing a possible bug with one external and two internal magnetometers. The used firmware is 1.8.2 (stable) with pixhawk 3 pro (by Drotek). External compass is DP0105 (NEO-M8N GPS + LIS3MDL Magnetometer for Pixhawk 3 Pro).
After calibration, I'm receiving CRITICAL error MAG2 SELFTEST CALIBRATION, and the drone is not able to arm. It seems that only way to arm is disabling CBRK_IO_SAFETY,  setting this parameter to 22027. Is it risky? Moreover, is it a bug, how could I proceed to overcome such a problem? I've repeated the calibration phase more and more, but the results were always the same.

I'm sorry, I've tried to upload flight log file in https://logs.px4.io/ but the following error is returned: HTTP Error 400: Failed to parse the file. It is most likely corrupt. I try attach it here,
[log_68_2019-5-26-20-53-32.zip](https://github.com/PX4/Firmware/files/3221272/log_68_2019-5-26-20-53-32.zip)

Below the screenshot:

[image]

Do you have any idea of the error, or suggestion to avoid the problem?
Do you have any evidence of a possible bug? 
Thank you so much!

Paul-DROTEK, 2019-05-27:
According to commits, this SELFTEST calibration step has been removed since firmware update 1.9.0.
You should git it a try.

codabat, 2019-05-27:
Hello Paul,
I also tried the stable 1.9.0. But I found problems on the GPS lock, and I get an invalid compass data.
I will try to update to stable 1.9.0 and repeat the calibration with more care. I'll let you know.
Thank you so much.
Best regards.

TaulPelu, 2019-05-27:
Ok then !
At work I performed the calibration process using the exact same device + Pixhawk3Pro + 1.9.0 firmware version and everything was working successfully. Hopefully you will be able to get the same results. :)

If there is still something wrong, maybe you should try doing the calibration while the autopilot + gps are not on board. Sometimes the drone frame can be a source of noise.

Paul-DROTEK, 2019-05-27:
Ok then !
At work I performed the calibration process using the exact same device + Pixhawk3Pro + 1.9.0 firmware version and everything was working successfully. Hopefully you will be able to get the same results. :)

If there is still something wrong, maybe you should try doing the calibration while the autopilot + gps are not on board. Sometimes the drone frame can be a source of noise.

codabat, 2019-05-27:
Dear Paul,
OK, I will reapet all calibation procedure tomorrow. I will inform you.
Thank you!

ch3sn3k, 2019-05-30:
Dear @codabat ,
we had a same issue. We had this error on at least three our units and it seems, that stable firmware 1.9.0 solved the issue.

Paul-DROTEK, 2019-05-30:
Hi,
Yes MAGSELFTEST has been removed since August 2018 in PX4 firmware version 1.9.0 alpha
see here : https://github.com/PX4/Firmware/commit/bb9c38a4803962c074ce0d49a57b64588f7b2a36#diff-aca721e521ad24ed19d30ebf3dd39374

codabat, 2019-06-01:
Hi @Paul-DROTEK, hi @ch3sn3k, hi all,
thanks for your feedback I updated the firmware to version 1.9.0.
I tried to recalibrate both without the eternal GPS on board.
After the calibration, I get the message "Internal compass inconsistent," and I am unable to arm the drone.
Moreover, I note that with the 1.9.0, the "GPS lock" occurs in very long times. The time required to carry out the "GPS lock" on same conditions seems to have returned to normal with the 1.9.0 master version (I refer to the daily May 31, 2019). I point out that the visibility conditions of the satellite constellation have always been excellent with 15-18 satellites visible from the "flight control".

Then I tried to update with the master version, I'm still receiving warnings, as  "The entry is denied..." and the message suggests enabling the safety switch.
Am I the only one to receive this error? Could the problem be related to the fact that Pixhawk is installed on the drone? However, I think it will always be a common situation.

I apologize in advance, I report some links with some collected logs, which maybe I can help you better understand what is happening to my flight controller.

[https://logs.px4.io/plot_app?log=d33870a5-1e54-474f-aea1-388258949072](url)
[https://logs.px4.io/plot_app?log=2f5ec665-0a3f-45e8-ba13-9042b490fea0](url) (v1.9.0 (ea48cd49))
[https://logs.px4.io/plot_app?log=b2f310be-831d-4eee-8b73-bfefff960de7](url) c8ea198a
[https://logs.px4.io/plot_app?log=0477d9b7-4607-4293-84b0-0fe2565a3788](url) c8ea198a

I noticed that the following trace is often shown in the logs:
[rm3100] no device on bus 4
Furthermore, but this is not of fundamental importance, you can tell me why if I download the logs through the telemetry they are merged but then after uploading the files the system tells me that they are corrupt. This is a possible bug?

I am still all to thank you very much. Your help is precious, and I am a bit perplexed and desperate right now.

Thank you again!

codabat, 2019-06-05:
Hello,
Sorry, no idea what causes the reported problems? Not even from the logs I included can you understand?

julianoes, 2019-09-04:
What's the status here? Has this been resolved?

codabat, 2019-09-04:
Yes, It's resolved.
Thank you.

closed 2019-09-04 by julianoes