PX4/PX4-Autopilot #12086: Can’t use pwm command to test AUX Pwm pins

Evaneee, 2019-05-27:
operating system: windows 7
board: pixhawk4

I need to test the IOpins and auxpins,I follow 
https://dev.px4.io/en/middleware/modules_command.html

**Question1:**
new start,I use
`pwm test -c 12 -p 1200`
I got this
[image]

 
It seems some controller press channel 1&2 to pwm 900, I don’t know why
I use
`mc_att_control stop`
`fw_att_control stop`
same result
but when After  I use  
`motor_ramp sine 1100 0.5`
result changed
[image]

seems works well, why this happened?

**Question2:**
I test auxpwm as IOpwm
Type
 `pwm test -c 12 -p 1200 -d /dev/pwm_output1`
But it does not work at all 
[image]
[image]

who can tell me why？
many thanks！！！

@julianoes Could you can give me some advise？？

julianoes, 2019-05-27:
@Evaneee thanks for the data. What version of PX4 are you using?

Evaneee, 2019-05-28:
> @Evaneee thanks for the data. What version of PX4 are you using?

Embarrassed，emmm， Could you tell me  what to do to  can get the PX4 version？

Evaneee, 2019-05-28:
> @Evaneee thanks for the data. What version of PX4 are you using?

is this??

[image]

julianoes, 2019-05-30:
Sorry, I did not yet get to this but it's on my list.

julianoes, 2019-06-05:
Question 1: That's interesting, I can reproduce this.

Question 2: The output of `/dev/pwm_output1` does not seem to change in the MAVLink output. I wonder if the data reported via MAVLink is wrong or if the PWM actually do not change. @Evaneee have you checked that?

julianoes, 2019-06-05:
@davids5 I'm looking into why `motor_ramp` works and `pwm test` does not for PX4IO.

I found that `motor_ramp` uses `PWM_SERVO_SET_ARM`, `PWM_SERVO_ARM`, `PWM_SERVO_SAFETY_OFF`:
https://github.com/PX4/Firmware/blob/d5298fccbd6714bed4ee9cac8579d31f99701de9/src/systemcmds/motor_ramp/motor_ramp.cpp#L295-L311

whereas `pwm` uses `PWM_SERVO_SET_MODE` -> `PWM_SERVO_ENTER_TEST_MODE`:
https://github.com/PX4/Firmware/blob/master/src/systemcmds/pwm/pwm.cpp#L658-L661

When looking at the PX4IO driver px4io.cpp it seems to me that this test mode is not implemented:
https://github.com/PX4/Firmware/blob/d5298fccbd6714bed4ee9cac8579d31f99701de9/src/drivers/px4io/px4io.cpp#L2749-L2751

For FMU in fmu.cpp it is however:
https://github.com/PX4/Firmware/blob/d5298fccbd6714bed4ee9cac8579d31f99701de9/src/drivers/px4fmu/fmu.cpp#L2131-L2133

Do you have an idea why this is the way it is? And do you know how it should be? :smile:

LorenzMeier, 2019-06-05:
From a design perspective there should not be a test mode directly in a driver (because of this). Tests should be applied externally. So the whole test mode should be stripped.

davids5, 2019-06-05:
PWM test - it was a change in the FMU that I needed the test mode flag for. The flag/mode just prevents the real mixer output from stepping on the test output.

julianoes, 2019-06-05:
I have no strong opinion but I would like have one way only, or if we really need both ways then both need to be implemented everywhere. We can discuss it in today's devcall.

Evaneee, 2019-06-19:
> Question 1: That's interesting, I can reproduce this.
> 
> Question 2: The output of `/dev/pwm_output1` does not seem to change in the MAVLink output. I wonder if the data reported via MAVLink is wrong or if the PWM actually do not change. @Evaneee have you checked that?

sorry I canot do the test, My /dev/pwm_output1 has disappeared, and I can not find it

for several days, I write a lot of printf to test the program,  and I find a little result about fmu
but suddenly I found it that I can not see pwm_output1 again........

https://github.com/PX4/Firmware/issues/12308

Evaneee, 2019-06-24:
> Question 1: That's interesting, I can reproduce this.
> 
> Question 2: The output of `/dev/pwm_output1` does not seem to change in the MAVLink output. I wonder if the data reported via MAVLink is wrong or if the PWM actually do not change. @Evaneee have you checked that?

you are right, I test it today, when do pwm test ,pwm_output1 can output ,but QGC do not show...

davids5, 2019-07-06:
Just to close the loop on this 
On HW with a px4io there will be 2 pwm devices `/dev/pwm_output0` and `/dev/pwm_output1`
Here are working test commands that I use to test withh (Used them this week)
```
fmu mode_pwm
pwm rate -m 0xff -g 0 -r 400 -d /dev/pwm_output1
pwm arm
pwm test -p 1234 -m 0xff
```

```
fmu mode_pwm
pwm rate -m 0xff -g 0 -r 400 -d /dev/pwm_output0
pwm arm
pwm test -p 500 -m 0x2 
```
The -m can b 1..128 in steps of 2: 1,2,4,8... and the will walk the pwm signal from ch1-ch7

Hope this helps.

I can not comment on weather QGC reports that PWM data back  - I use a oscilloscope to measure HW as it reduces the chain of errors to a minimum: Probe on the right pin :)

bozkurthan, 2019-07-16:
Hello all, 
I have similar problem with this Issue. I use generic X quadrotor with latest build.
I removed aux cables from Pixhawk to test something. After it I forgot  aux cable numbers' order and just wanted to test each channel with this code `pwm test -p 1000 -c 1` (-c 2 3 4 also) to decide which one is correct plugged.

Before start I just connected USB to pixhawk and closed USB circuit breaker. I didn't close mc_pos_control. Also pressed safety switch.
When I call `pwm test -p 1000 -c 1` command, motor1 is started and everything was normal. But when I aborted test and tried to change channel with `pwm test -p 1000 -c 2` , only calibration sound comes once and after it nothing happens when I try to call command several times.  Channels don't respond my command. Each channel respond with calibration sound when I first call command for its. After it nothing happens when I tried to call command several times. I can only test channel 1.

 After whole steps, I cut power and started with `pwm test -p 1000 -c 2` command. And channel 2 responds my commands while others not. 

Finally, I wanted to try this command at the first, `pwm test -p 1000 -c 1234` and after this command I could test each channel separately. (-c 1 -c 2 or -c 12). Moreover, when I started  `pwm test -p 1000 -c 12` with this command, channel 3 and 4 don't respond pwm test command. Only once calibration sound are coming.

I just wanted to dig into this problem and I see that `pwm steps -c 1234` command solves this problem. When I called this command calibration sounds come for each channel except channel that I used this `pwm test -p XXX -c X` command. And I can try the pwm test command for each channel. 

Whole process should be like this?

davids5, 2019-07-20:
@bozkurthan - you have to set the rate on all the channels you want to test first. 

`pwm rate -m 0xff -g 0 -r 400 -d /dev/pwm_output1`

Once that is done, the channels are in PWM mode and can be tested.

bozkurthan, 2019-07-21:
Ok, I see. It's mentioned on documentation. Thanks again @davids5

closed 2019-10-25 by julianoes