PX4/PX4-Autopilot #12087: BMI055 Gyro driver uses "Reserved register" for Gyro Temp readout

ksschwabe, 2019-05-27:
In the BMI055 gyro driver, it seems from the code that the author wanted to read out the Accel Temp register (see https://github.com/PX4/Firmware/blob/master/src/drivers/imu/bmi055/BMI055_gyro.cpp#L552). But because the Gyro part of chip is a separate SPI block, with a separate chip select and interface, the Gyro driver ends up reading out register 0x08 from the GYRO block of the BMI055. This register is "reserved".

See BMI055 datasheet (https://ae-bst.resource.bosch.com/media/_tech/media/datasheets/BST-BMI055-DS000.pdf) pg. 50 for the ACCD_TEMP register 0x08 and pg. 97 for the *Reserved* GYR register 0x08

junwoo091400, 2022-11-22:
Code location changed: https://github.com/PX4/PX4-Autopilot/tree/main/src/drivers/imu/bosch/bmi055

@ksschwabe would you be able to check on current code, whether this still exists?

ksschwabe, 2022-11-23:
@junwoo091400 It seems to me that the new driver is no longer reading out the temperature from the "Reserved" register. So, this resolves my initial concern. However I can't see how the temperature for the "gyro device" is now being set, so that you can perform temperature compensation on the data. Which is peculiar, because gyros exhibit rather strong temperature drift.