PX4/PX4-Autopilot #12091: position mode through mocap

roseyangyu, 2019-05-28:
I use 1.8.2/1.9.0 firmware, v2 and v5 flight control. LPE is not used. I want to locate through MoCap. The accuracy of the mocap raw data is millimeter level. I sent the position to flight control via VISION_POSITION_ESTIMATE. Vehicle can enter position mode at rest, but when vehicle fly up, it can't hover accurately. I looked at Local_position's data with a QGC. The x,y axis occasionally mutates to more than 10 meters, with a greater change in height, generally changing at dozens of meters, and occasionally larger to more than 100 meters. How can I debug to make the plane hover steadily?

julianoes, 2019-08-26:
You need to add a logfile, otherwise it's hard to see what's going on.

julianoes, 2019-08-27:
Ok thanks for the answer. Sounds like the issue is resolved and I can close it. Please re-open if the issue comes back.

closed 2019-08-27 by julianoes

roseyangyu, 2019-08-27:
There was no unlocking during testing because local_position displayed unreliable data.But I change mavlink_message.I send mocap rather than vision.Then I adjust mocap coordinate system to NED and turn PID.Vehicle can position well.

roseyangyu, 2019-08-28:
OK,Thank you for your attention