PX4/PX4-Autopilot #12092: Bug: Clearing mission not possible

AuterionWrikeBot, 2019-05-28:
Clearing mission was not possible.
We had the same issue last week. QGC tries it several times and gives out error message.
Should be easy to reproduce on a bench.

AuterionWrikeBot, 2019-05-28:
➤ Mischa Zihler commented:

Thomas Gubler who would be responsible on the PX4 side to solve this?

AuterionWrikeBot, 2019-05-28:
➤ Julian Oes commented:

Thanks for the note. I will try to reproduce this.

AuterionWrikeBot, 2019-05-29:
➤ Julian Oes commented:

I can't reproduce this in SITL with QGC.

AuterionWrikeBot, 2019-06-28:
➤ Mischa Zihler commented:

Julian Oes we had that problem again with yesterday's testing. we were able to clear mission at the first time, after connection has been established. but after creating a new mission and flying it, we couldn't clear it anymore. do you have more information about this?

AuterionWrikeBot, 2019-06-30:
➤ Julian Oes commented:

Mischa Zihler is the issue reproducible, so does it happen everytime? And if it happens, do other actions still work such as setting a parameter, etc. If that doesn't work either it points to a general communication problem. And when it happens, what is the actual error message?

AuterionWrikeBot, 2019-07-01:
➤ Mischa Zihler commented:

Julian Oes
UAV: IF750-010
Videolink/Telemetry: Microhard
Groundstation: Android

I think it is reproducable. It always happened after we flown a mission. I don't think the general communication was a problem, because we could remove the waypoint manually, recreate a mission, upload and fly it. See the error message as attachment.

AuterionWrikeBot, 2019-07-01:
➤ Julian Oes commented:

Ok thank you. I will try to reproduce it by flying a mission in-between then.

AuterionWrikeBot, 2019-07-01:
➤ Julian Oes commented:

I can't reproduce it in SITL or HITL.

AuterionWrikeBot, 2019-07-01:
➤ Mischa Zihler commented:

Julian Oes Today it worked of course. We were able to clear it everytime. I'll inform you, when it happens again

dagar, 2019-07-01:
The log might help if you manage to reproduce it while the logger is running.

Can you reshare the image directly on github?

julianoes, 2019-07-02:
[image]

hamishwillee, 2019-10-01:
@julianoes Can we still verify? The image above does not look like QGC

julianoes, 2019-10-01:
@hamishwillee I was never able to reporoduce this.

closed 2019-10-01 by julianoes