PX4/PX4-Autopilot #12093: How to see the px4_debug(…) information

Evaneee, 2019-05-28:
Found it that when do pwm test, the program will execute 
D:\PX4\home\Firmware\src\systemcmds\pwm\pwm.cpp ,and in this file DO
	`int fd = px4_open(dev, 0);`
	`PX4_INFO("dev ============:%s",dev);`
	`PX4_INFO("dev's fd========:%d",fd); `
Adding two lines above to see the value of dev and fd, I got this picture, The dev is changed but fd value is 3 all the time,…
[image]
trace it, go to \PX4\home\Firmware\src\lib\cdev\posix\cdev_platform.cpp
Add one line `PX4_INFO("Do px4_opennnnnnnnnnn"); ` before  `PX4_DEBUG("px4_open");`
Build and upload the program
But I cant get any information in nsh mavlink console about px4_open 

And I don’t want to Add PX4_INFO() every time 
How to see the PX4_DEBUG() info??
[image]

roseyangyu, 2019-05-31:
You can add "printf("px4_open")" and you can get it through serial4/5

Evaneee, 2019-06-03:
I follow the https://dev.px4.io/en/debug/system_console.html 
and want to connect to system console

Connect to the fmu debug port using usb2serial cable
and set putty like below, when power on pixhawk4 ,I can get some "!@#$!@#$!@#$@$@#$........."In the console
I dont know it means what 

Can you help me????

julianoes, 2019-09-03:
The file descriptor `fd` is just a number that `open` returns (per NuttX task). It has nohas no global meaning. So it makes sense that it's always 3 for your case.

To enable PX4_DEBUG for a module, I usually do something like this at the beginning of the file:

```
#undef PX4_DEBUG
#define PX4_DEBUG PX4_INFO
```

closed 2019-09-03 by julianoes