PX4/PX4-Autopilot #12094: build px4 above 1.80 does not work anymore macbook pro Sierra 10.12.6

taileron, 2019-05-28:
**Describe the bug**
A clear and concise description of the bug.

**To Reproduce**
Steps to reproduce the behavior:
1. Drone switched on '...'
2. Uploaded mission '....' (attach QGC mission file)
3. Took off '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/en/getting_started/flight_reporting.html)).
- Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/).

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- Describe the type of drone.
- Photo of the IMU / autopilot setup if possible.

**Additional context**
Add any other context about the problem here.

taileron, 2019-05-28:
up to 1.80 make still works perfect for versions 1.82 and above the following error occurs:Failed to import yaml.
You may need to install it with 'sudo pip install pyyaml'

Traceback (most recent call last):
  File "/Users/xxxxx/src/Firmware/Tools/serial/generate_config.py", line 13, in <module>
    import yaml
ImportError: No module named yaml
[13/1215] Copying NuttX/nuttx to build/px4_fmu-v4_default/NuttX
ninja: build stopped: subcommand failed.
make: *** [px4_fmu-v4_default] Error 1
(tried to install pyyaml and retried the whole homebrew procedure)

LorenzMeier, 2019-05-28:
You need to install the dependencies according to the dev guide.

LorenzMeier, 2019-05-28:
https://dev.px4.io/en/setup/dev_env.html

closed 2019-05-28 by LorenzMeier

taileron, 2019-05-28:
that I have done before ... all expect pypaml displays already installed.

src xxxxx$ sudo -H pip install pyserial empy toml numpy pandas jinja2 pyyaml
Password:
Requirement already satisfied: pyserial in /Library/Python/2.7/site-packages
Requirement already satisfied: empy in /Library/Python/2.7/site-packages
Requirement already satisfied: toml in /Library/Python/2.7/site-packages
Requirement already satisfied: numpy in /System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python
Requirement already satisfied: pandas in /Library/Python/2.7/site-packages
Requirement already satisfied: jinja2 in /Library/Python/2.7/site-packages
Collecting pyyaml
  Could not fetch URL https://pypi.python.org/simple/pyyaml/: There was a problem confirming the ssl certificate: [SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:590) - skipping
  Could not find a version that satisfies the requirement pyyaml (from versions: )
No matching distribution found for pyyaml

taileron, 2019-06-06:
Attention .... 1.8.2 doesn´t built with GCC 7.3.1 anymore which gets installed by the current  dependencies
1.8.2 needs GCC 7.2.1 then it works flawlessly with my Macbook Pro Sierra
1.9.0 needs GCC 7.3.1

julianoes, 2019-06-07:
@taileron regarding the pyyaml install failure, I googled this for you:
https://github.com/pypa/pip/issues/5226