PX4/PX4-Autopilot #12095: make px4_fmu-v4_default

oasinic, 2019-05-28:
Hi all
I use Ubuntu 18.04 to build the code(v1.8.2) 
Here is the input and out put:
make px4_fmu-v4_default
-- PX4 config file: /home/mikhail/PX4/Firmware/boards/px4/fmu-v4/default.cmake
-- PX4 config: px4_fmu-v4_default
-- PX4 platform: nuttx
-- PX4 lockstep: disabled
-- PX4 version: v1.9.0-rc1-26-g9d8015d029
-- cmake build type: MinSizeRel
-- The CXX compiler identification is GNU 7.2.1
-- The C compiler identification is GNU 7.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /home/mikhail/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc
-- Check for working CXX compiler: /home/mikhail/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-g++
-- Check for working CXX compiler: /home/mikhail/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /home/mikhail/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc
-- Check for working C compiler: /home/mikhail/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PythonInterp: /usr/bin/python (found version "2.7.15") 
-- Found PY_jinja2: /usr/local/lib/python2.7/dist-packages/jinja2  
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1
-- Using C++11
-- Release build type: MinSizeRel
CMake Error at cmake/px4_add_module.cmake:210 (get_target_property):
  get_target_property() called with non-existent target
  "uavcan_generic_driver".
Call Stack (most recent call first):
  src/drivers/uavcan/CMakeLists.txt:70 (px4_add_module)

-- ROMFS: px4fmu_common
-- ROMFS: Adding rc.board_defaults
-- ROMFS: Adding rc.board_sensors
-- ROMFS: Adding rc.board_extras
-- Configuring incomplete, errors occurred!
See also "/home/mikhail/PX4/Firmware/build/px4_fmu-v4_default/CMakeFiles/CMakeOutput.log".
See also "/home/mikhail/PX4/Firmware/build/px4_fmu-v4_default/CMakeFiles/CMakeError.log".
Error: /home/mikhail/PX4/Firmware/build/px4_fmu-v4_default is not a directory
Makefile:193: recipe for target 'px4_fmu-v4_default' failed
make: *** [px4_fmu-v4_default] Error 1

in "/home/mikhail/PX4/Firmware/build/px4_fmu-v4_default/CMakeFiles/CMakeError.log".

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: /home/mikhail/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-g++ 
Build flags: 
Id flags:  

The output was:
1
/home/mikhail/gcc-arm-none-eabi-7-2017-q4-major/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/lib/libc.a(lib_a-exit.o): In function `exit':
exit.c:(.text.exit+0x2c): undefined reference to `_exit'
collect2: error: ld returned 1 exit status

Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: /home/mikhail/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc 
Build flags: 
Id flags:  

The output was:
1
/home/mikhail/gcc-arm-none-eabi-7-2017-q4-major/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/lib/libc.a(lib_a-exit.o): In function `exit':
exit.c:(.text.exit+0x2c): undefined reference to `_exit'
collect2: error: ld returned 1 exit status

Thanks!

julianoes, 2019-05-28:
Can you try:
```
make clean
git submodule update --init --recursive
make px4fmu-v4_default
```

oasinic, 2019-05-29:
```
make clean
mikhail@mikhail-pc:~/PX4/Firmware$ git submodule update --init --recursive
Submodule 'boards/atlflight/cmake_hexagon' (https://github.com/PX4/cmake_hexagon.git) registered for path 'boards/atlflight/cmake_hexagon'
Submodule 'cmake/configs/uavcan_board_ident' (https://github.com/PX4/uavcan_board_ident.git) registered for path 'cmake/configs/uavcan_board_ident'
Submodule 'src/modules/micrortps_bridge/micro-CDR' (https://github.com/PX4/micro-CDR.git) registered for path 'src/modules/micrortps_bridge/micro-CDR'
Cloning into '/home/mikhail/PX4/Firmware/boards/atlflight/cmake_hexagon'...
Cloning into '/home/mikhail/PX4/Firmware/cmake/configs/uavcan_board_ident'...
Cloning into '/home/mikhail/PX4/Firmware/src/modules/micrortps_bridge/micro-CDR'...
Submodule path 'boards/atlflight/cmake_hexagon': checked out '08fd0a73045346448adf6969660196228b23e1fa'
Submodule path 'cmake/configs/uavcan_board_ident': checked out '2e5f9d6768b1dbffc006dc2ceeb2bfe120f22163'
fatal: Needed a single revision
Unable to find current revision in submodule path 'platforms/nuttx/NuttX/nuttx'
```

julianoes, 2019-05-29:
> but i install nuttx using script

I don't understand why or how you did that.

oasinic, 2019-05-29:
use faq  https://dev.px4.io/en/setup/dev_env_linux.html

```
Pixhawk/NuttX (and jMAVSim)
To install the development toolchain:

Download ubuntu_sim_nuttx.sh.
Run the script in a bash shell:
source ubuntu_sim_nuttx.sh
You may need to acknowledge some prompts as the script progresses.
Restart the computer on completion.
```

julianoes, 2019-05-29:
Ah ok, that should all be correct.
Can you try `make submodulesclean`, this should remove all submodules and clone them again fresh.

oasinic, 2019-05-30:
everything worked, thx!

closed 2019-05-30 by oasinic

mcelhennyi, 2019-10-15:
I am having a similar outcome here. The above suggestions did not help.

I am on Ubuntu 16.04 in a docker container.

I am trying to run `make px4_fmu-v3_default`. I started from a fresh repo, from tag v1.9.2. I have previously ran the nuttx shell script mentioned above (with a reboot after). Additionally I am building from within a docker container. 

I also tried from my own private version of the px4 repo. This version I can build on ubuntu and it runs just fine. 

My target is the Pixhawk RadioLink hardware.

When I first ran the make command I was prompted with the "type u and hit enter" for the submodules, so I ran through all that, then it got to the end with this error:

```
> autogenerated headers for rev https://github.com/mavlink/mavlink/tree/fc20e8f5192738a9f1adca0e1f8d96f71c54ce3c

 *******************************************************************************
 *   IF YOU DID NOT CHANGE THIS FILE (OR YOU DON'T KNOW WHAT A SUBMODULE IS):  *
 *   Hit 'u' and <ENTER> to update ALL submodules and resolve this.            *
 *   (performs git submodule sync --recursive                                  *
 *    and git submodule update --init --recursive )                            *
 *******************************************************************************

   Only for EXPERTS:
   mavlink/include/mavlink/v2.0 submodule is not in the recommended version.
   Hit 'y' and <ENTER> to continue the build with this version. Hit <ENTER> to resolve manually.
   Use git add mavlink/include/mavlink/v2.0 && git commit -m 'Updated mavlink/include/mavlink/v2.0' to choose this version (careful!)

u
Synchronizing submodule url for 'mavlink/include/mavlink/v2.0'
Submodule path 'mavlink/include/mavlink/v2.0': checked out '57000134021fb8d5c2c6281e3783ce29bb35bc17'
Submodule path 'mavlink/include/mavlink/v2.0': checked out '57000134021fb8d5c2c6281e3783ce29bb35bc17'
Submodule fixed, continuing build..
-- ROMFS: px4fmu_common
-- ROMFS: Adding rc.board_defaults
-- ROMFS: Adding rc.board_sensors
-- Configuring incomplete, errors occurred!
See also "/home/user/gazebo_dev/px4-firmware/build/px4_fmu-v3_default/CMakeFiles/CMakeOutput.log".
See also "/home/user/gazebo_dev/px4-firmware/build/px4_fmu-v3_default/CMakeFiles/CMakeError.log".
Error: /home/user/gazebo_dev/px4-firmware/build/px4_fmu-v3_default is not a directory
Makefile:193: recipe for target 'px4_fmu-v3_default' failed
make: *** [px4_fmu-v3_default] Error 1

```

hamishwillee, 2019-10-15:
@mcelhennyi Don't necrobump. Re-post with your issues. IN this case it is most likely that you haven't cleaned your submodules. You might do something like:
```
make distclean
git submodule update --recursive
make px4_fmu-v3_default
```