PX4/PX4-Autopilot #12105: EKF2: Position lost after EV stops (GPS + EV Fusion) - 1.9 stable

TobiMiller, 2019-05-29:
Hey,

actually i want to fuse the pose information of my zed camera with gps into the px4. After stopping the vision the base_link is not moving anymore and jumps back to map at position (0,0,baro_hgt). It doesnt give any local_position from GPS data.  I thought that it is a real fusion and will rely on gps when vision is gone. see this post https://github.com/PX4/Firmware/issues/10062.
At start without vision the local_position is working on GPS.

my setup is following:

- zed camera publishs pose message referenced to its origin frame zed_origin.
- mavros uses map->vision_estimate lookup instead of topic /mavros/vision_pose/pose.
- i publish static map->zed_origin transform to get zed pose into ENU.
- i publish zed_origin->vision_estimate from zed pose msg

map --> zed_origin --> vision_estimate (my solution)
map --> base_link (published by mavros)

mhkabir, 2019-05-29:
Please post a log.

TobiMiller, 2019-06-04:
ok. on my next fieldtest.

mhkabir, 2019-06-04:
@TobiMiller please make it a log from boot (set SDLOG_MODE to "from boot until shutdown") so that we can replay it. Thanks!

closed 2019-09-16 by stale[bot]