PX4/PX4-Autopilot #12115: Ask for a further improvement of a TO-DO about offboard mode

SalimTerryLi, 2019-05-30:
For general questions please use [PX4 Discuss](http://discuss.px4.io/) or [Slack](http://slack.px4.io/).

**Describe problem solved by the proposed feature**
This [TODO](https://github.com/PX4/Firmware/blob/a18f3e5d45e73ce7afd4988bed2e0e92ecad6b59/src/modules/mavlink/mavlink_receiver.cpp#L1369) prevents the offboard mode working on a rover. The expected solution should not limit the offboard mode as multirotor-only.

**Describe your preferred solution**
I' m asking for adding `att_sp.thrust_body[0] = set_attitude_target.thrust;` below but I don' t know how it would affect other frame types.
This will fix that rover cannot get throttle value from attitude setpoints.

**Describe possible alternatives**
I don' t know if there is any other way to solve this.

julianoes, 2019-05-30:
@SalimTerryLi thanks for raising this! Have you already tried to make this work by changing it? If so you could make a pull request that we could discuss.

julianoes, 2019-05-30:
FYI @ItsTimmy

Jaeyoung-Lim, 2019-05-30:
It would be awesome if this fix leads to enabling offboard mode in fixed wing too

SalimTerryLi, 2019-06-03:
#12149

closed 2019-07-17 by SalimTerryLi