PX4/PX4-Autopilot #12130: Telemetry Uplink not working if sd card disconnected accidentally

caomuqing, 2019-05-31:
**Describe the bug**
A clear and concise description of the bug.
A flight plan is pre-planned and uploaded to the vehicle. Before flight, sd card reads ok. During auto mission, the SD card accidentally lost connection(not sure why), then the telemetry uplink also does not function anymore, meaning whatever command sent from QGC will not be executed by the UAV at all(an error will be prompted). However, the downlink is still ok, meaning the uav data is still available in real time in QGC.

**To Reproduce**
Steps to reproduce the behavior:
1. simply turn on the drone on ground
2. disconnect the sd card while autopilot is on
3. try to send command from QGC
4. the error appears

bkueng, 2019-06-03:
Thanks for the report. Can you be more specific on the commands you send and errors that you see? Which PX4 version is that?

A quick test in HIL confirms there are issues when ejecting the SD card during a mission: the `read` syscall [here](https://github.com/PX4/Firmware/blob/master/src/modules/dataman/dataman.cpp#L629) did not return anymore (I tested on a Pixracer and Pixhawk 4). Consequently RTL does not work anymore, but I can still switch modes via QGC and take control via RC.
@davids5 can you have a look?

caomuqing, 2019-06-05:
I have tried changing mode command, mission upload command, both does not work. the PX4 version is 1.8.2.
[image]
[image]

bkueng, 2019-06-06:
Can you try with v1.9.0? Which hardware do you use?
In which order did you send the commands? Can you try only a mode switch after disconnecting the SD card?

closed 2019-09-04 by julianoes