PX4/PX4-Autopilot #12138: Infinitely flying up during RTL

anandbaskaran, 2019-06-01:
We have been flying mission mode with the stable release 1.8.2 of PX4 (with Pixhawk 4: fmu-v4) between two points that are 700m away. We used RTK GPS, and a barometer for position and altitude estimation. The drone took off (to the altitude of 25m) and started towards the next waypoint. 

A few seconds later, the manual pilot engaged `RTL Home` and the drone started to climb up with full throttle. Due to fast climb and far distance the communication with RC pilot was lost. Fast discharge of battery with full throttle engaged safe land mode. However, the drone did not land in the home position. Unfortunately, we were not able to retrieve logs from the drone’s autopilot. 

[Just the drone status logs were recovered](https://github.com/PX4/Firmware/files/3240429/dronelogs.recovered.txt)
`NOTE:` RTL_RETURN_ALT param was set to 15.0m. 

-------------------------------------------

To understand what happened similar tests have been conducted in a simulation. And at times we have observed comparable behavior as described in the following screenshots:
[image]

When the RTL Home is engaged the drone goes up to the altitude of 30m (instead of 15m). After drone reaches the home waypoint, the drone loses its GPS and lands randomly. 
[image]

Please find the corresponding status logs below: 
[image]

This issue of GPS loss is obtained once in a while during the simulation and we could not identify the way of repeatability. However, flying back at 30m (instead of 15m happens almost always). In reality, the drone flew at least above 100m.

jkflying, 2019-06-04:
The landing position issue is fixed here, I believe: https://github.com/PX4/Firmware/pull/12032

However I'm not sure about the altitude of RTL issue.

julianoes, 2019-07-09:
Can you check these params?
https://dev.px4.io/en/advanced/parameter_reference.html#RTL_RETURN_ALT
https://dev.px4.io/en/advanced/parameter_reference.html#RTL_DESCEND_ALT

julianoes, 2019-08-07:
@anandbaskaran has this ever happened again? And if yes, could you upload a log?
I don't know how to reproduce this.

anandbaskaran, 2019-10-02:
@julianoes It did not happen again. However, we also started using v1.9.0, which had a new issue that it automatically engages kill switch. I will post it along with the logs.

julianoes, 2019-10-02:
> new issue that it automatically engages kill switch

Ok please create a new issue for this. I'll close this issue, you can re-open it if it happens again.

closed 2019-10-02 by julianoes