PX4/PX4-Autopilot #12152: v1.9.0 Release. Max pwm on main outputs in stabilize mode with throttle stick zero.

group174311, 2019-06-03:
In stabilized mode, on actuators outputs the maximum control signal was applied, when throttle was in zero.

**To Reproduce**
Steps to reproduce the behavior:
1. arm drone.

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=cf2378c0-ad28-4927-a5ec-c1afa80af0bf

bresch, 2019-06-04:
Hi @group174311 , I found out that you set the parameters `PWM_MAIN_REV[1-2]`. I guess you wanted to reverse the rotation of the rotors. The effect of this parameter is to reverse a servo an not a motor as it inverts the PPM output: 1000us becomes 2000us, this is why the motors 1 and 2 went full throttle.
In order to reverse the direction of a motor, simply swap two phases.
I improved the description of the parameter in this PR (https://github.com/PX4/Firmware/pull/12165) to avoid confusion.

group174311, 2019-06-04:
@bresch thanks a lot. Our mistake, we whant to invert the servos for the elevons. We will double check next time.

closed 2019-06-04 by group174311