PX4/PX4-Autopilot #12158: VTOL airframes don't start up

sfuhrer, 2019-06-03:
After flashing master on a pixhawk4 and selecting a VTOL frame (eg standard vtol, autostart 13000) it doesn't start up cleanly (eg it is no able to connect to QGC). Other airframes work (eg 2100 or 4001).

It seems to be something mixer related, as the start up works (get connection to QGC) if I skip the mixer loading via "set MIXER skip" and "set MIXER_AUX skip" in the config file.

@dagar @RomanBapst @MaEtUgR

dagar, 2019-06-03:
Do you recall the last time this specific configuration (airframe + board) worked?

Shot in the dark, could you try https://github.com/PX4/Firmware/issues/12158 quickly?

sfuhrer, 2019-06-03:
Well with 1.9 it for example works, didn't check anything more recent.

And yes, with #12158 it works also fine.

Tony3dr, 2019-06-05:
@dagar we are having similar issues with the Convergence, Pixhawk 4 mini. The telemetry on the vehicle is not sending the data, no connection on the Master branch. The last time we flew the unit was on PR [12096](https://github.com/PX4/Firmware/pull/12096). We just recently got the unit up and running.

MaEtUgR, 2019-06-07:
I'm currently this happens because NuttX hits the maximum number of 32 tasks and everything after that is not started anymore. We had the problem today on a multicopter with multiple MAVLink instances as well. If you look at the startup in the console you see all commands failing after a certain point.

@dagar I think you already know this problem exists.

dagar, 2019-06-10:
Can you try https://github.com/PX4/Firmware/pull/12230?

sfuhrer, 2019-06-11:
I've directly tried current master, as I've seen that #12230 is already merged. And yes, VTOL now works again.

dagar, 2019-06-11:
Fixed in https://github.com/PX4/Firmware/pull/12230.

closed 2019-06-11 by dagar

dannyfpv, 2019-06-13:
I Tested master  on Pixhawk4 mini V5 vtol 
Modes Tested 

Position Mode: no issue 
Altitude Mode: no issue 
Stabilized Mode: no issue 
Mission Plan Mode: no issue 
-RTL: no issue 
transsition mode: no issue

log:
https://review.px4.io/plot_app?log=16fc215a-ab05-4e05-9313-fbf05f54e633