PX4/PX4-Autopilot #12168: Switching to mission blocked until land/disarm/arm

jkflying, 2019-06-04:
Currently, the drone can be armed and flown in stabilized, altitude, position etc without having a full GPS lock. This then leads to the drone not being able to switch to mission mode _even after_ it has taken off. In order to switch to mission mode, it must first land, disarm, and then arm again. This is not very user friendly, and the error message is not very clear - "Reject Auto Mission" - which doesn't help in solving the issue. See eg. https://github.com/PX4/Firmware/issues/11741

Potential solutions, in rough order of safest to least safe:
- Don't allow the drone to arm until it has a GPS lock sufficient to satisfy mission, if any of the position-aware modes might be switched to during flight.
- Do whatever resets are required in-flight to allow the switch to mission (set the takeoff location to first GPS lock location?)
- Allow the mission to specify the RTL location, rather than using the takeoff location
- Deprecate this requirement of having the home location in order to switch to mission, and only require it for switching to RTL. I understand this is done from a safety perspective, to allow RTL, but RTL can be triggered from any mode, not just auto modes, so I don't see why only they would require the RTL location to be set.

LorenzMeier, 2019-07-03:
This is likely related to the `mission_feasibility_checker.cpp` file and this section in it:

```
	// first check if we have a valid position
	const bool home_valid = _navigator->home_position_valid();
	const bool home_alt_valid = _navigator->home_alt_valid();

	if (!home_alt_valid) {
		failed = true;
		warned = true;
		mavlink_log_info(_navigator->get_mavlink_log_pub(), "Not yet ready for mission, no position lock.");

	} else {
		failed = failed || !checkDistanceToFirstWaypoint(mission, max_distance_to_1st_waypoint);
	}
```

jkflying, 2019-07-03:
I think this is actually a duplicate of https://github.com/PX4/Firmware/issues/12204