PX4/PX4-Autopilot #12169: [RTK] Localization problem as soon as RTK is fixed

TobiMiller, 2019-06-04:
Hey @Stifael @bkueng 
i need a better and stable localization. that's the reason why i jumped from this issue https://github.com/PX4/Firmware/issues/12105 to RTK.

I have a Drotek TinyRTK Module on my UAV connected to GPS port (has no magnetometer). I have a second normal NEO 8 GPS Module connected to Telem2 and the Magnetometer to I2C. GPS2 is deactivated (i want the magnetometer). I am getting a good normal gps reception. while moving around by hand i see the movement in QGroundcontrol. 

I have also a TinyRTK connected to QGroundcontrol which sends RTCM to my UAV. When my UAV is on RTK (float) i can see the movement in QGC. BUT! When it goes to RTK (fix) then nothing happens. 

here the log:
1st movement -> just gps
2nd movement -> rtk (float)
3nd "movement" -> rtk (fix)
https://review.px4.io/plot_app?plots=pid_analysis&log=ee5e58a5-2662-441f-9143-c3b348291b9b

i dont know if it is related to this issue:
https://github.com/PX4/Firmware/issues/11716

bkueng, 2019-06-05:
The log shows it never goes into fixed RTK mode, but falls out of RTK for the 3rd movement.
What link do you use for the RTCM data?

TobiMiller, 2019-06-05:
jep. that is what i have seen today. it goes from 3D RTK (float) to 3D DGPS Lock.
HDOP and VDOP jump from under 1.0 to 100.0

i am using some Hobbyking 915mhz telemetry modules. and i send RTCM over Qgroundcontrol (the autoconf way). Do the PX4 stack and QGroundControl setup the RTK modules correct from every actual config?

bkueng, 2019-06-06:
There can be several reasons for dropping out of RTK mode:
- the telemetry bandwidth is too low / link quality decreases when moving. You can monitor `gps status` to check that the RTCM injection rate stays stable.
- GPS signal is quality is low. You might get RTK when holding still, but as you move you lose it (I've seen this before)

> Do the PX4 stack and QGroundControl setup the RTK modules correct from every actual config?

Yes

TobiMiller, 2019-06-07:
found the problem. it was a faulty cable with loose contact on plug side. strange because it was a retail cable and is looking fine. i think the RX on GPS side was loose and hasnt get the RTCM messages and config.

closed 2019-06-07 by TobiMiller