PX4/PX4-Autopilot #12180: Quadrotor keeps jumping when flying in altitude mode

parakhm95, 2019-06-04:
#### Description
The problem occurs in altitude mode whether I switch in-flight or before takeoff. When the robot is taking off, it keep gaining altitude and doesn't come down. If you lower the stick to zero, it keeps climbing but starts jumping in the air. It falls for a few milliseconds and then gains thrust. It keeps repeating this while continuously gaining altitude. Same thing happens if you are flying in manual and you switch to altitude mode.

#### Reproduction
1. Switch to altitude mode.
2. Arm.
3. Bring throttle to take-off position.
4. Robot keeps climbing, so you lower the stick to control but nothing happens.
5. Lower the stick further down to zero, the robot start to **jump**.
6. Retake manual control.

**Expected behavior**
It should just takeoff and hold altitude. 

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=5ef531fd-d7db-443b-8d76-c330044f0b29
You will clearly see how even when the thrust is at it's lowest position, there are huge spikes in the thrust of the robot as seen in actuator controls. 

**Drone**
The drone is a 165mm quadrotor with 3750kV motors and 3040 Triblade props. It is using a Pixfalcon(also called Pixhawk Mini). The power is supplied by the PDB-ESC Combo. 

**Additional context**
I have used this same setup with Ardupilot stack and it works fine somehow.

bresch, 2019-06-11:
Hi @parakhm95 , 
It looks that you forgot to adjust the hover throttle. Since your vehicle has a higher thrust/weight ratio than usual, you should reduce the hover throttle to about 0.3 (from what I see in your log). The parameter is [MPC_THR_HOVER](https://dev.px4.io/master/en/advanced/parameter_reference.html#MPC_THR_HOVER).
As shown in the picture below, the land detector "thinks" the drone is touching the ground since the throttle is low compared to the defined hover throttle. It switches in and out of that "ground contact" state and this is why the throttle is jumping around.
[image]

parakhm95, 2019-06-11:
I tried the fix you suggested and yes it works! Thank you so much. This has been stuck for a very long time now. 

Also, how did you plot these flags? Are they available on the log review online and need to be enabled or do I have to manually find them in the log? 

But it works for sure.

bresch, 2019-06-12:
@parakhm95 I'm glad it worked out! Those flags are not shown on FlightReview (online tool) since you don't need to check them on a regular basis. To see them, you need to open the file using a program such as [PlotJuggler](https://docs.px4.io/master/en/log/flight_log_analysis.html#plotjuggler).

closed 2019-06-12 by bresch