PX4/PX4-Autopilot #12190: land_detector complains about timeout

julianoes, 2019-06-05:
Recently with SITL I see this error output on startup:

```

ERROR [land_detector] Timed out while waiting for thread to start
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 14570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
...
```

To reproduce:
```
make px4_sitl jmavsim
```

@dagar any idea?

dagar, 2019-06-05:
Recently as in today, or earlier?

dagar, 2019-06-06:
Can you try https://github.com/PX4/Firmware/pull/12196?

mcsauder, 2019-06-11:
Hi @julianoes  and @dagar ,I think I have it fixed in PR #12209 at the schedule call [here](https://github.com/PX4/Firmware/blob/8ecb925c3687a46d6f1aa91ac7d0d384658a6a4e/src/modules/land_detector/LandDetector.cpp#L68).  Let me know if that works for you.

mcsauder, 2019-06-11:
One last thought, changing from the present `ScheduleOnInterval()` call to `ScheduleNow()` preserves the previous behavior prior to the general work queue, so this should be a safe change when it is accomplished: https://github.com/PX4/Firmware/blob/53aa4130a8cc549887106b077aae2f77bcacf662/src/modules/land_detector/LandDetector.cpp#L69

mcsauder, 2019-06-11:
@dagar , adding a `ScheduleNow()` call allows the `ModuleBase::wait_until_running()` call to succeed.

This means that `_object.load()` doesn't pass unless `ScheduleNow()` gets called.  Do you see what's going on there?

PR #12196 (2019-06-16): land_detector initiate cycle immediately
 - fixes #12190
commit 106ee280 (2019-06-16): land_detector initiate cycle immediately

- fixes #12190
closed 2019-06-16 by dagar