PX4/PX4-Autopilot #12202: Multi-Vehicle simulation Error

randyd55, 2019-06-06:
I have been attempting to run through this multi-vehicle simulation tutorial found here https://dev.px4.io/en/simulation/multi-vehicle-simulation.html
When I run the launch command, Gazebo runs and spawns the vehicles correctly, but in the terminal I get the errors
`Error binding socket` 
and
`Failed to connect with daemon: Permission Denied`
I suspect that the errors stem from the fact that the tutorial assumes an earlier version of the px4 Firmware. For instance, later in the tutorial it mentions editing startup files for the vehicles that don't seem to exist in the latest version.

Any ideas on how I can successfully run through this tutorial with the latest version of the Firmware?

julianoes, 2019-06-07:
@randyd55 have you ever run SITL with `sudo`? Can you check if there are any files in `/tmp/` named `px4-*` and check using `ls -l` if they are owned by `root`? That would explain the issue you're seeing and would mean you have to delete these files (using `sudo`).

randyd55, 2019-06-07:
So I managed to get rid of the errors by doing what you said, but still no px4 prompt appears in the terminal

julianoes, 2019-06-08:
I think in multi vehicle mode the command prompts are disabled because how would you select which vehicle you're interacting with. Instead you can use things like MAVROS/MAVSDK/QGC to interact with the drones.

closed 2019-06-08 by julianoes