PX4/PX4-Autopilot #12203: High RAM usage on Pixhawk 2.1 Cube

DanielePettenuzzo, 2019-06-06:
We have noticed that in the last few weeks the memory usage on the Cube has increased by over 7kB and now with the same setup we had working, we already run out of RAM at boot. 

@sfuhrer can share the two commits from master that we tested. One from last week and one from a couple weeks ago. Please also share a log so that we can reproduce the same exact setup (in terms of parameters).

From the top of my head I can say that we were running VTOL (so both multicopter and fixed wind modules) and we had two MAVLink instances running, one of which was TELEM2 with high rates going to the companion computer. 

Is this RAM increase expected and if yes how can we decrease it so that we can still fly VTOL with the cube?

DanielePettenuzzo, 2019-06-06:
FYI @RomanBapst

dagar, 2019-06-06:
I'll take a look at your log once it's posted, but you're likely seeing the memory cost of the new WQ and the corresponding thread pre bus. This was expected and it's why [uORB changes](https://github.com/PX4/Firmware/pull/12040) were pursued at roughly the same time, which will win back a lot more memory (hopefully soon).

If anyone would like to help with the transition please test and review PRs related to uORB::Subscription and the new WQ.

 - https://github.com/PX4/Firmware/pull/12123 <- most important
 - https://github.com/PX4/Firmware/pull/12176
 - https://github.com/PX4/Firmware/pull/12186
 - https://github.com/PX4/Firmware/issues/12068
   - Once everything is moved out of Nuttx HPWORK and LPWORK we can disable them in most configurations and save another few kB.

sfuhrer, 2019-06-07:
Here's the log: https://review.px4.io/plot_app?log=06617924-5fd8-4142-8617-2d0d8a18eba6
It was with master from last Friday, 43e3fc707de823f. RAM is at 99%. 

Before we were flying with https://github.com/sfuhrer/Firmware/tree/Babyshark (close to master in mid April). We already had RAM at around 95%, so the RAM difference is about 10kB.

dagar, 2019-06-07:
This is the other effort that has the potential to save a large amount of memory soon. 
https://github.com/PX4/Firmware/pull/12207

In particular for the VTOL case I believe we could have all 5 controllers (mc_att_control, mc_pos_control, fw_att_control, fw_pos_ctrl_l1, vtol_att_control) that are currently separate tasks moved to run out of 2 WQ threads. This will save something like 10-15 kB (completely separate from the savings mentioned above in https://github.com/PX4/Firmware/issues/12203#issuecomment-499703008) and slightly reduce cpu usage.

julianoes, 2019-09-06:
@DanielePettenuzzo still an issue?

sfuhrer, 2019-09-06:
> @DanielePettenuzzo still an issue?

No, with the recent WQ changes not an issue anymore.

closed 2019-09-06 by sfuhrer