PX4/PX4-Autopilot #12204: Global position is valid but still no home position is set at takeoff

DanielePettenuzzo, 2019-06-06:
I had a couple flights in the last weeks where I did a takeoff in position mode and then tried to switch to mission using a switch on the rc and the mission was rejected. After looking at the log I noticed that it was rejected because no home position was set. The strange thing is that during takeoff my global position was already valid so I don't understand why my home position was not set. Anyone has any insights on why this might be happening?

Here is one of the logs I found to have this issue:
https://review.px4.io/plot_app?log=0974afe4-dadb-4a84-8633-7de483f4f0f8

I was flying a racer frame with a pixracer with both GPS and optical flow. The branch just had some minor unrelated things and was based on v1.9.0-beta.

DanielePettenuzzo, 2019-06-06:
FYI @nicovanduijn

nicovanduijn, 2019-06-07:
@ItsTimmy may have some additional logs of the same behavior occurring on the rover

julianoes, 2019-06-07:
It looks like the GPS accuracy needs to be good enough for home to be set. Global position valid is not enough on its own:

https://github.com/PX4/Firmware/blob/be02ad3514213af7e1febd0377835eb00d063495/src/modules/commander/Commander.cpp#L1099-L1100

bresch, 2020-04-01:
[image]
All the required conditions were good: eph, epv, condition_global_position_valid, condition_local_position so the only one that could have failed is the home position publisher.

We didn't see the issue since then so I this this got fixed. I'm closing it now, we can still reopen if needed.

closed 2020-04-01 by bresch