PX4/PX4-Autopilot #12206: Add position lock to resolve quadcopter sinking in offboard mode

shrit, 2019-06-07:
**Describe the bug**
I have noticed that the IRIS quadcopter have tendency to lose altitude in gazebo when pitching or rolling in some direction. The problem have been reported and well discussed
on Slack you can follow using this thread:
https://px4.slack.com/archives/C68J8H32A/p1559816218007600 

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=97830f7e-5bce-4649-8704-e5d0ffc8ea5b

julianoes, 2019-06-07:
Since (free) Slack deletes all conversation after some time, let me give some more context to this.

This is using [velocity offboard control in MAVSDK](https://github.com/mavlink/MAVSDK/blob/22fcf4b8deade56e679f8580d035b9457aac7f97/src/plugins/offboard/include/plugins/offboard/offboard.h#L187-L192). The problem is that the drone will try to hold the z velocity but does not correct drift over time.

A possible fix would be to lock the z position if z velocity is set to 0.

I had done something like that a while ago before position control received flight tasks:
https://github.com/PX4/Firmware/pull/5868/commits/642e65cffc5a7199d3847a09bf06814b1f153a90

MaEtUgR, 2019-06-07:
I'm shifting towards having reusable logic of the flight tasks in libraries anyways and we can just reuse the position lock from manual position mode which locks the altitude/position when the velocity is constantly zero. The start for this library can be found here: https://github.com/PX4/Firmware/pull/12072/files#diff-cd11905871652d5f1467db1dfc9cecd2R47

dakejahl, 2019-06-09:
Nice! Can we have this perform braking? It would be idea if  `<0,0,0>` velocity setpoints caused the vehicle to brake and then lock into a position hold after coming to a stop. 

I am imagining the aggressiveness of the braking could be controlled via a parameter.

shrit, 2019-07-17:
Just to add some more details: After monitoring several flights of the quadcopter. I have noticed that sometimes the quadcopter do not just sink, but gain some altitude. For example: a quadcopter might gain up to 5 meters up when it is going forward, and then it starts to sink gradually.

shrit, 2020-01-15:
Keep open

shrit, 2020-04-14:
Keep open

Pedro-Roque, 2020-05-02:
@MaEtUgR is this still ongoing?

shrit, 2020-05-02:
@Pedro-Roque Yes, it still open.

shrit, 2020-08-01:
Keep open

Jaeyoung-Lim, 2020-08-01:
@shrit Why would you use attitude setpoints if you want to control the attitude? Why is using velocity setpoints / position setpoints not an option?

shrit, 2020-08-01:
@Jaeyoung-Lim  I use only velocity setpoints. In this case, I was using the velocity setpoints to set the velocity of quadrotor every 50 milliseconds (In order to simulate the behavior of a joystick for example)

shrit, 2021-06-07:
Keep open