PX4/PX4-Autopilot #12208: here2 GNSS gps compatible with Firmware v.1.8.0

jbourne15, 2019-06-07:
Are the magnetometers in the Here2 GNNS gps compatible with px4 firmware v1.8.0?

dagar, 2019-06-10:
Please try v1.9.0. Support for this magnetometer was added post v1.8.0.

dagar, 2019-06-10:
@hamishwillee worth capturing in docs when support for each thing appeared?

closed 2019-06-10 by dagar

hamishwillee, 2019-06-10:
@dagar That happens automatically - when support is added for a new product, a corresponding page must be added in the user guide to master (and will be copied whenever a new version is released). So if a product is not in the guide in a version you can assume it is not supported in that version.

It is the responsibility of whoever adds support to a board to make sure this page is created. I can help with that, but I need info on the board, links to ownership of board, instructions on how to connect it. 

This broke down in this case - I have been waiting on @profincnc to help create a page https://github.com/PX4/px4_user_guide/issues/474

From now on let's just create a minimal placeholder page that states at least that we support the product. Done here: https://github.com/PX4/px4_user_guide/pull/502