PX4/PX4-Autopilot #12216: NuttX STM32F7 write-back dcache

dagar, 2019-06-08:
Due to ARM 1259864 Data corruption eratta we currently have the dcache disabled on stm32f7. One solution is to switch to write-back data cache configuration, but this is currently broken for other reasons (starting at our own px4io driver https://github.com/PX4/Firmware/pull/12075).

 - https://nuttx.org/doku.php?id=wiki:howtos:port-drivers_stm32f7#write-back_vs_write-through_d-cache

LorenzMeier, 2019-06-08:
@davids5 Is there a quick path to get there?

davids5, 2019-06-08:
I had spent a little bit of time on this before discussing it with Daniel. It's really just a matter of spending the time to ferret out all of the issues. The obvious ones that hard faults will be the first. Second Step would be figuring out the subtle bugs due to misalignment

closed 2019-07-10 by dagar