PX4/PX4-Autopilot #12217: Add buzzer for OmnibusF4 board

robomaniacpl, 2019-06-08:
**Describe problem solved by the proposed feature**
Actually buzzer connected to OmnibusF4 board doesn't make sound, so for that reason I'd like to add buzzer/tune_control functionality to omnibusf4 board.

**Describe your preferred solution**
The buzzer is connected to PB4 pin on STM32F405RGT6. I did some changes to make it works but unfortunately I'm not able to make any sound from buzzer. All my changes below.
```
diff --git a/boards/omnibus/f4sd/default.cmake b/boards/omnibus/f4sd/default.cmake
index 25a970e081..136b0bbf8b 100644
--- a/boards/omnibus/f4sd/default.cmake
+++ b/boards/omnibus/f4sd/default.cmake
@@ -37,11 +37,12 @@ px4_add_board(
                rc_input
                stm32
                stm32/adc
-               #stm32/tone_alarm
+               stm32/tone_alarm
                #tap_esc
                #telemetry # all available telemetry drivers
                telemetry/frsky_telemetry
                #test_ppm
+               tone_alarm
                osd
 
        MODULES
diff --git a/boards/omnibus/f4sd/nuttx-config/nsh/defconfig b/boards/omnibus/f4sd/nuttx-config/nsh/defconfig
index 25def0f8cd..b219c4f7c8 100644
--- a/boards/omnibus/f4sd/nuttx-config/nsh/defconfig
+++ b/boards/omnibus/f4sd/nuttx-config/nsh/defconfig
@@ -160,7 +160,7 @@ CONFIG_STM32_SPI1=y
 CONFIG_STM32_SPI2=y
 CONFIG_STM32_SPI3=y
 CONFIG_STM32_TIM1=y
-CONFIG_STM32_TIM3=y
+CONFIG_STM32_TIM3=n
 CONFIG_STM32_TIM5=y
 CONFIG_STM32_UART4=y
 CONFIG_STM32_USART1=y
diff --git a/boards/omnibus/f4sd/src/board_config.h b/boards/omnibus/f4sd/src/board_config.h
index 4fc43f6ee3..d0c02b17f4 100644
--- a/boards/omnibus/f4sd/src/board_config.h
+++ b/boards/omnibus/f4sd/src/board_config.h
@@ -193,6 +193,12 @@
 #define PX4_I2C_BUS_EXPANSION    2
 #define PX4_I2C_BUS_LED          PX4_I2C_BUS_EXPANSION
 
+/* Tone alarm output */
+#define TONE_ALARM_TIMER       3       /* timer 3 */
+#define TONE_ALARM_CHANNEL     1       /* channel 1 */
+#define GPIO_TONE_ALARM_IDLE   (GPIO_OUTPUT|GPIO_PUSHPULL|GPIO_SPEED_2MHz|GPIO_OUTPUT_CLEAR|GPIO_PORTB|GPIO_PIN4)
+#define GPIO_TONE_ALARM                (GPIO_ALT|GPIO_AF2|GPIO_SPEED_2MHz|GPIO_PUSHPULL|GPIO_PORTB|GPIO_PIN4)
+
 /* PWM
  *
  * 6 PWM outputs are configured.
```
Annotation.
Without 
```
CONFIG_STM32_TIM3=n
```
I get the following error 
```
error: #error Must not set CONFIG_STM32_TIM3 when TONE_ALARM_TIMER is 3
 #  error Must not set CONFIG_STM32_TIM3 when TONE_ALARM_TIMER is 3
```

Do you have any  idea how to make it working?

dagar, 2019-06-10:
@bkueng any interest in upstream buzzer support for the omnibus?

bkueng, 2019-06-11:
If anyone creates a PR, I'll review and test, otherwise I prefer buzzer via d-shot support.

closed 2019-09-23 by stale[bot]