PX4/PX4-Autopilot #12221: State estimation sensor problems to typhoon_h480 model in sitl

jannsta1, 2019-06-09:
**The bug**

Since Firmware Commit https://github.com/PX4/Firmware/commit/43e3fc707de823faa19b16091e5ee4a900e3c446 the typhoon_h480 model does not exhibit good position control in gazebo simulation with mavros. Still the case as at https://github.com/PX4/Firmware/commit/b0176dc88aecd22363f771bfe62dafa57a6a194a. The error message ```Critical navigation failure! Check sensor calibration``` is logged in conjunction with the problem.

From release v1.9.0 to the commit prior to the reported issue (4a4d323) seem to work fine.

**To Reproduce**
Steps to reproduce the behavior:
1. Load commit version and ensure a totally clean build/initialisation (problem may not manifest unless all these steps are followed):
```
cd <FIRMWARE LOCATION>
git checkout <COMMIT UNDER TEST>
make clean
git submodule update
rm ~/.ros/eeprom/parameters
make px4_sitl_default gazebo_typhoon_h480
```
2. roslaunch model ```roslaunch px4 mavros_posix_sitl.launch vehicle:=typhoon_h480```
3. Arm / transition to offboard mode
4. Send position target

**Expected behavior**
Stable position control behaviour (from commit 4a4d323):
https://logs.px4.io/plot_app?log=6c788665-95f7-4b2b-a963-9ad58911024c

**Log Files and Screenshots**
Unstable position control behaviour:
https://logs.px4.io/plot_app?log=894c8553-18b1-4fc1-ad4a-d1587485637e

**Additional context**
- Iris model still seems to work OK in this regard
- Have also observed ```Preflight Fail: Compass #0 uncalibrated``` but possibly due to not clearing parameter files between different build levels.
- when using ```commander takeoff``` in gazebo via the pxh> shell the vehicle seems stable.

jannsta1, 2019-06-17:
Issue still present as far as I can see - would be great to see the reason this has been closed?

FlavioTonelli, 2019-06-17:
You have probably been misled by the referenced Weekly Digest.
This issue is still open

jannsta1, 2019-06-17:
So it is - silly me!

Jaeyoung-Lim, 2019-06-22:
@jannsta1 Do you still have this issue? I can fly the H520 in SITL with the current master

jannsta1, 2019-06-22:
@Jaeyoung-Lim - nice chatting with you at the dev summit :) and thanks for looking into this. I last checked with the master 5 days ago and still saw the issue - won't have access to my workstation until Monday so will try then and report back. Just to check though - is there a H520 model in the sitl_gazebo repo as well as the H480? Its also worth mentioning that its important to do the clean build steps (step 1 from the initial bug report) to ensure that everything has been rebuilt with desired git checkout.

jannsta1, 2019-06-24:
@Jaeyoung-Lim - update as promised - I am still seeing the issue on latest master. Couple of other notes: 

1) It is still possible to fly the vehicle - just that following takeoff the z position control is very poor and the 
warning messages reported in the bug report are present.
2) x & y position control seem to work as normal
3) The z position control stabilises eventually, but a new z position setpoint causes high oscillations again (I am using barometer for height control)

Finally - looking at the changes made on the commit when I started seeing this (listed above), there were many changes made to the sitl sensor IDs - I just didn't notice anything that would explain why this would effect the typhoon model and not the iris models.

Jaeyoung-Lim, 2019-06-27:
@jannsta1 I will look into this as soon as possible. Currently I am overloaded with other stuff not related to SITL, so won't be able to look at it this week

Jaeyoung-Lim, 2019-10-11:
I have tested this and it can fly without any problems. @jannsta1 Please reopen if the problem persists

closed 2019-10-11 by Jaeyoung-Lim