PX4/PX4-Autopilot #12235: Bypass accel check in preflight

tuloski, 2019-06-10:
Hi,
I'd like to bypass the preflight check for the accelerometers.
In particular I'd like to bypass this check: https://github.com/PX4/Firmware/blob/bef7a9ba8e9382a1497cf35cfe2f9a4b5ff97a28/src/modules/commander/PreflightCheck.cpp#L270-L281.
Only for testing I commented all those lines and compiled and checked in SITL, but I still get "Accel Range, hold still on arming" when arming.
I'm confused because I don't understand where that error is coming since I commented it and it seems the only place where it might happen.

Also is there any big drawback if arming and taking off with large accelerations (>15 m/s/s)? Was it ever tested in real life without this check and with large accelerations (for example in a drifting car)?

dagar, 2019-06-10:
It catches completely bogus sensor setups early on. All of these checks should be broken out into a bitmask parameter within commander.

For your local build try cleaning first and rebuilding. Otherwise share the branch on github and I can take a quick look.

closed 2019-06-10 by dagar

tuloski, 2019-06-12:
The code is the master with simply those lines commented. I really can't understand. It seems like there is another code that perform the same check. 
Other parts modified of PreflightCheck.cpp are compiled correctly and they work.

dagar, 2019-06-12:
There are also reminders to hold still in the calibration routines, but you'll only see that during calibration. Beyond that you'll need to share more detail (screenshots, code, etc) for me to be able to tell you what's actually happening.

tuloski, 2019-06-12:
My fault: I noticed that when running with "make px4_sitl_default" it was working while it wasn't working with ros .launch file.
I solved by building with catkin_make also the px4-firmware package in my catkin workspace.
I'm not sure what it was happening, but probably not finding the compiled "px4" package it was going to look somewhere else for the executable, not getting the ones compiled with "make".

fabrizioschiano, 2021-01-08:
Is there a way to bypass this preflight check without having to comment PX4 code? I am trying to do this also in a SITL simulation environment. 
There are other preflight checks that one can disable such as `FD_FAIL_R` ([here](https://docs.px4.io/master/en/advanced_config/parameter_reference.html#FD_FAIL_R)) 
Thanks.