PX4/PX4-Autopilot #12237: Air data module extension to multiple sensors

AuterionWrikeBot, 2019-06-11:
□ enable multiple airspeed sensors (in sensors and AirspeedModule)
 □ Calibration procedure with multiple sensors?
 □ simulate different airspeed failures (clogging, water, ..) and test detection and switching

hamishwillee, 2020-02-09:
@RomanBapst So is this something we will still want? It was you originally, then @sfuhrer. Good to assign or close.

RomanBapst, 2020-02-10:
@hamishwillee This is not done yet. Wen can already use 2 airspeed sensors but we are not there yet in terms of calibration. What we've done so far is that we used a sensirion airspeed sensor as the second sensor, because it does not require any calibration.
I think @dagar is really the man who has been doing most progress in this direction with all this sensor pipeline refactoring.

hamishwillee, 2020-02-10:
That's cool @RomanBapst . My point was really that if this has no assignee, then what is likely to happen is it will go stale and get dropped. If that is going to happen anyway, let's just delete it now.

So I have assigned to @dagar and you guys can do with it as you will.

Safranek42, 2020-08-25:
> @hamishwillee This is not done yet. Wen can already use 2 airspeed sensors but we are not there yet in terms of calibration. 

Is this feature of using two airspeed sensors included in the current master version? Or it will be just along with the calibration feature?
I connected two airspeed sensors (ms4525 pitot tube sensors) via i2c, and only one is visible. I have checked the number_of_airspeed_sensors variable in the Firmware, and it is always one. Also in the logfile there is only one airspeed sensor logged. MAX_NUM_AIRSPEED_SENSORS is set to 3. Should it work with no setup or I need to set something in the code/parameters to see the second sensor?

Thanks for any help.

hamishwillee, 2020-08-26:
@Safranek42 Unlikely this has been updated. @sfuhrer can you provide any update?

saengphet, 2020-08-26:
I have tested this feature by connecting 2 airspeed sensors with different brand "SDP33" and "ms4525". Only SDP33 was found as primary airspeed. I investigated more and found that only 1 differential_pressure from SDP33 pass on to airspeed module to calculate indicated airspeed and more. I tried to modified code in sensors.cpp to make it works.

 However, due to the concept of measuring the pressure of "SDP33" and "ms4525" is completely different, then offset pressure from these 2 is too much difference too, I could not make it works perfectly (sometimes, SDP33 is primary but not always).
 The important thing is Qgroundcontrol does not support 2 airspeed sensors calibration and offset pressure has only 1, not 2.

For the same airspeed sensor brand such as "ms4525", I could not find a multi-instance to call them at the same time (might be due to there is the same i2c address??).

mlaiacker, 2020-10-02:
Hi,
I just also got a similar problem in v1.11 and I am questioning the airspeed_selector module idea. 
It compares airspeed but there is only one instance that publish airspeed topic in "sensors.cpp" (search for "ORB_ID(airspeed)") 
You can have multiple differential pressure sensors but the "sensors.cpp"  only looks at one "ORB_ID(differential_pressure)" and doesn't convert all "ORB_ID(differential_pressure)" to "ORB_ID(airspeed)". Why??
This makes no sens at all. 

Also I always have a differential_pressure instance from the analog "ADC_AIRSPEED_VOLTAGE_CHANNEL" even if the "diff_pres_analog_scale" is set to zero. I have to remove the "ADC_AIRSPEED_VOLTAGE_CHANNEL" from the board_config.h to make a sdp3x work. Otherwise I get the temperature from the sdp3x in the airspeed topic but no airspeed. 
Makes no sens at all.

[image]

Any ideas?

dagar, 2020-10-02:
@mlaiacker it's in flux at the moment, but if you're interested in full support and can help test I think we could get this done properly. In https://github.com/PX4/Firmware/pull/15853 I'm refactoring the airspeed portion of `sensors` so that it either selects the highest priority differential pressure to publish a single airspeed, or you can optionally have it publish an airspeed for each differential pressure.

The analog airspeed publishing is clearly a bug. If you open a new issue and share a log I'll take a closer look.

mlaiacker, 2020-10-02:
@dagar 
Thank you for the explanation. I was confused how the airspeed_selector should work. It looks like i was right that it is not finished. That's ok.

The analog sensor thing:

Looks like a bug I introduced myself by some changes I made to the Firmware. I tested the original v1.11 and it worked fine. 
Sorry about that. 
I had to introduce a gain/scaling factor on the differential pressure level to compensate for not optimal pitot tube placement on the airframe and somehow the gain was zero. I don't know how, but that my job to find out. 

Maybe this scale factor for the diff pressure could be added to PX4?

dagar, 2020-10-02:
> Maybe this scale factor for the diff pressure could be added to PX4?

I'm going to expand the differential pressure calibration to be more like accel/gyro/mag where it's per device id. We can add a scale factor as well.