PX4/PX4-Autopilot #12240: Tiltrotor motors shaking

julianoes, 2019-06-11:
When disarmed, the motors of the tiltrotor shake in tiltrotor SITL.

To reproduce:
```
make px4_sitl gazebo_tiltrotor
```

The shaking goes away when you do:
```
commander arm
```
but re-appears after:
```
commander disarm
```

FYI @sfuhrer 

@dlwalter raised the issue in https://discuss.px4.io/t/gazebo-tiltrotor-model-joints-are-shaking-in-1-9-0/11355/2.

dlwalter, 2019-06-11:
I'm guessing this is related to reducing the real-time update rate from 800 to 250 to implement the lock step in the gazebo_mavlink_interface.

dlwalter, 2019-06-11:
Can confirm, disabling lockstep in the model.sdf and increasing the real-time update rate to 800 (max step size of 0.00125) fixes the issue.

julianoes, 2019-06-11:
@dlwalter I would like to understand why or how that fixes it. I can see that it might prevent the visible shaking but there still seems to be something going on.

dlwalter, 2019-06-11:
I agree - I just needed a workaround for now.  Looking forward to an update.

julianoes, 2019-06-12:
@RomanBapst do you have any idea here?

jlecoeur, 2019-06-12:
I hit the same issue a while ago : my model's hinges were completely unstable. If I remember correctly, setting a larger number of iterations fixed it.

julianoes, 2019-09-24:
Ok, to me this sounds like we need to enable higher real time update rates again and make them work with lockstep.

sfuhrer, 2019-12-24:
Yes, disabling lockstep and setting real_time_update_rate to 500 and  max_step_size to 0.002 helped to get rid of motor shaking in disarmed state and made the flight much more stable (especially the transitions were very unstable before). Has anybody an idea what the root cause could be?

RicardoM17, 2020-01-16:
This is fixed by https://github.com/PX4/sitl_gazebo/pull/386 no? Or are we still looking for the root cause and a more global fix?

P.s. I didn't test it, just linking the issues/PRs

sfuhrer, 2020-01-17:
For me it fixed the shaking of the tilt, but not really lead to a more stable flight (mainly transitions). There only setting real_time_update_rate to 500 and max_step_size to 0.002 solved it. Wondering if it's also the case for other users. 
But this is actually another problem then, as this issue was originally just about the shaking.

julianoes, 2020-02-12:
@sfuhrer how can you "see" the more stable flight with higher update rate?

sfuhrer, 2020-02-14:
> @sfuhrer how can you "see" the more stable flight with higher update rate?

This is with the default settings (lockstep enabled, rate at 250 and step at 0.004): 
[image]
And this with lockstep disabled, rate 500 and step 0.002:
[image]

LorenzMeier, 2020-02-14:
Shouldn’t we be able to just ramp up the sample rate also in lockstep?

julianoes, 2020-02-14:
> Shouldn’t we be able to just ramp up the sample rate also in lockstep?

That might "just" work. I'll try it and see what else depends on this rate. It will likely increase CPU and decrease the max possible sim speedup that is possible.

julianoes, 2020-02-14:
The "just" is not so straightforward because if we change the default update rate it means that all models in all submodules for Gazebo and jMAVSim, would have to be changed, so it's quite a bit of churn. The more long term solution would be to support varying sample rates while still staying in lockstep.

Jaeyoung-Lim, 2020-02-14:
@julianoes I think this is something occuring commonly with position controlled joints (also observed on the gimbal on the h480. I will try to see if I can tune the joint control better

julianoes, 2020-02-14:
@sfuhrer it would be good if you could test tiltrotor with this wip branch:
https://github.com/PX4/Firmware/tree/pr-increase-rate (make sure to also update the sitl_gazebo submodule).

It raises the rate from 250 Hz to 500 Hz. I want to confirm if this actually fixes the issue or if the issue is something about lockstep and not the rate itself.

Btw. iris speedup in Gazebo decreases from 25x to 11x when doubling the rate. Presumably, we could optimize this with optimizations like pipelining the messages, so we don't have to wait on the roundtrip of messages each time  but can free-wheel to a certain extent.

julianoes, 2020-02-14:
@LorenzMeier ok this turns out to be a bit more complex. Doubling the rate is trivial for multicopter which run at the gyro rate. For VTOL, it falls apart right after the transition when we switch to fixedwing control which runs at the ekf2 rate, so probably 250 Hz.

I'll think about potential ways to work around this.

julianoes, 2020-02-17:
@sfuhrer ok please fetch this branch again to test. Gazebo now runs at 500 Hz but everything with PX4 is only transmitted at 250 Hz.

sfuhrer, 2020-02-17:
> @sfuhrer ok please fetch this branch again to test. Gazebo now runs at 500 Hz but everything with PX4 is only transmitted at 250 Hz.

Yes it now looks good :+1: :+1:

julianoes, 2020-02-19:
PR is done: https://github.com/PX4/sitl_gazebo/pull/415

sfuhrer, 2020-02-27:
With PX4/sitl_gazebo#415 the tiltrotor sitl problems (both shaking and unsteady transitions) seem to be solved, so I'm gonna close this.

closed 2020-02-27 by sfuhrer