PX4/PX4-Autopilot #12241: Offboard attitude rate control broken

julianoes, 2019-06-11:
There seems to be a regression for offboard control since 1.9. Starting offboard control, so switching the mode to offboard control does no longer work.

Offboard control gets rejected:
```
[03:37:30|Warn ] command temporarily rejected (176). (mavlink_commands.cpp:168)
/home/julianoes/src/MAVSDK/src/integration_tests/offboard_attitude.cpp:83: Failure
Expected equality of these values:
  offboard->start()
    Which is: 4-byte object <05-00 00-00>
  Offboard::Result::SUCCESS
    Which is: 4-byte object <01-00 00-00>
```

**To Reproduce**

Run the MAVSDK integration test against jMAVSim SITL:
```
cmake -Bbuild -H. && cmake --build build -j8 && build/src/integration_tests/integration_tests_runner --gtest_filter="SitlTest.OffboardAttitudeRate"
```

julianoes, 2019-06-11:
`git bisect` points to 3faab909d7db25bef6dc3e2eb753cdc05f7f9ce5!

dagar, 2019-06-11:
No, but I'll look into it. We should really get these tests running against PX4/Firmware directly.

dagar, 2019-06-11:
What are the general requirements of offboard control mode? Is it expected that you start streaming some type of offboard setpoint prior to changing modes?

julianoes, 2019-06-11:
> What are the general requirements of offboard control mode? Is it expected that you start streaming some type of offboard setpoint prior to changing modes?

Exactly, you need to have an offboard setpoint set before you can switch to the mode.

commit 6816f2a4 (2019-06-13): commander force offboard control update when first entering mode

- fixes #12241
closed 2019-06-13 by dagar