PX4/PX4-Autopilot #12247: EKF2 not merging the Rangefinder

JMshare, 2019-06-11:
**Describe the bug**
The estimated altitude and the local_position.z seem to ignore the rangefinder data.
Using the Lidar Lite through PWM,
Pixhawk 4 fmu_v5,
stable release 1.9.0,
in position control or altitude hold mode.

The distance sensor data show that the sensor outputs correct readings but the altitude estimates seem uninfluenced.

**Log Files and Screenshots**
https://review.px4.io/plot_app?log=d19babaf-06b4-4074-b14e-af8828fb51c2

Add screenshots to help explain your problem.
[image]

gajena, 2019-07-18:
@JMshare Did you find any solution for this? I am also having the same problem with fmu_v3 and LidarLite.

JMshare, 2019-07-18:
No, I don't really know how to fix this, hoped someone from the developers would react.

But if it helps to track the issue, the reported _**signal_quality 0**_ in the log must be the reason why the ekf ignores the data. Also I remember somehow coming to a conclusion that this is related to LidarLite specifically, not just any rangefinder. The newest model had some changes in the firmware that haven't been accounted for in the PX4 yet or something like that...

gajena, 2019-07-18:
I have tested the same LiDaR with v1.8.1, and it was working fine. So after digging in the firmware, I found the heck.https://github.com/AerialRobotics-IITK/Firmware/commit/25e0d01e3a869651d2cb21ba1477ae0c30fee030

JMshare, 2019-07-19:
Thanks @gajena, I was thinking about doing the same thing until the driver is fixed.

tecdee, 2019-09-03:
@gajena sorry if the question is silly but i'm quite new to px4 and all, so how do I make use of the hack?
- Modify the 'ekf2_main.cpp' file as per the hack from the link you gave
- build the firmware, and load as custom firmware.
Am I right ?
Thanks for your help.

gajena, 2019-09-04:
Yes or just pull from my repo(please pull latest commit).

tecdee, 2019-09-04:
> Yes or just pull from my repo(please pull latest commit).

Thank you @gajena