PX4/PX4-Autopilot #12251: PX4 1.9.0 ignores RC-commands

hardeman, 2019-06-12:
**Describe the bug**
We're testing with a Pixhawk with PX4 1.9.0. The problem is since the upgrade to 1.9.0.

The RC is connected and stick movements are visible in QGroundControl, but PX4 doesn't respond to arming or a mode switch sometimes. After a couple of restarts it works just fine.

**To Reproduce**
Steps to reproduce the behavior:
-Everything switched on. 
-In QGroundControl are stick movements visible.
-The commands (like arming or mode switch) are not executed

dagar, 2019-06-12:
Can you share a log showing the problem? This definitely isn't reproducible in general (RC is tested constantly).

hardeman, 2019-06-13:
There is no log showing the problem. The drone can not be armed, can not switch to another mode, etc.

But the movement of the sticks are visible in QGroundControl. 

The most I can show is a video when it's working and a video where it isn't.

dagar, 2019-06-13:
You can get a log from boot by setting the parameter SDLOG_MODE to 2.

HVKHVK, 2019-06-13:
Hi there,

I definitely have the same problem, and that happened just as he said. After I upgraded my v1.8.2 stable to v1.9.0, my device started to respond slowly for arming like arming 2 seconds after I pull the stick and disarming almost 4 seconds later than the stick.  I did arming and disarming several times, and it did not solve. So I unplug the battery and replug it several times and problem solved. I also saw the stick movements on Qground as he said. I am uploading several logs, but I am not sure about their capabilities. I tried to recreate the bug, but I failed. Here the three following logs of the system, which I believe contains the bug. 

https://review.px4.io/plot_app?log=a7c08c8a-a018-47fc-a6dd-5b85665dfacc
https://review.px4.io/plot_app?log=9c83ed28-6056-486d-8b3b-0bb99ef2dbd9
https://review.px4.io/plot_app?log=12c942df-f99a-4f57-89bc-3a242c0c1d3e

Thanks

dagar, 2019-06-13:
@PX4/testflights have you seen anything like this?

hardeman, 2019-06-13:
It is not happening everytime, so I will upload the log when it happens. Thank you so far.

dagar, 2019-06-13:
@hardeman could you describe your setup? Specific RC equipment, etc.

Junkim3DR, 2019-06-13:
> @PX4/testflights have you seen anything like this?

@dagar we have not seen this issue so far, we will comment a log with the vehicle behavior later on today.

Junkim3DR, 2019-06-13:
### Tested on Pixhwak 4 Mini v5

**Modes tested:**
- Position Mode: Good. 
- Altitude Mode: Good. 
- Stabilized Mode: Good. 
- Mission Plan Mode (Automated): Good. 

**Procedure**
Armed and took off in every mode, to determine if vehicle responded as expected. 

**Notes**
Logged from boot until shutdown, vehicle did not present the bug mentioned.

**Logs**
https://review.px4.io/plot_app?log=763da36f-015a-40bb-83b6-793faaa7d3ae

### Tested on Pixhawk 4 v5

 **Modes Tested**
- Position Mode: Good.
- Altitude Mode: Good.
- Stabilized Mode: Good.
- Mission Plan Mode (Automated): Good.
- RTL (Return To Land): Good.

**Procedure**
Arm and Take off in position mode, after flying for approximately one minute, switched to altitude then stabilized mode proceed to switch to mission plan mode then make sure that vehicle follows all waypoints as shown in QGC, once completed all waypoint then trigger RTL and see landing behavior.

**Notes**
No issues noted, good flight in general.

**Log**
https://review.px4.io/plot_app?log=a8f5509e-1de5-4f6f-8d31-d0bc5302c450&fbclid=IwAR0q_6Kfg7R7MyFZqJTD4EmzDwzT7rZx37pLRV_DIZEhwuF62B_PeYjU90w

JMshare, 2019-06-16:
Same thing with the arming or kill switch happens to me sometimes. 
My feeling is that it has something to do with how I start or terminate the flight/vehicle.

**For example**, if a landing detector disarms the airframe but my RC stick is still in the armed position, the commander would seem to deal with it by ignoring the RC stick ever after until reboot. Same thing if I arm using QGC slider but my stick is still in disarmed mode. And lastly if I accidentally power on the airframe before turning on the transmitter, the RC arming commands would get ignored even though in QGC I can see them switching in the Flight modes setup window. 

**Disclaimer**
These are just general hints of what could lead to the ignored RC switches, I am not sure if I am seeing these consistently, basically I try to avoid the situations that would get me into the described positions.

**Optimal behavior** would be to reset the priority to RC after the commander detects alignment of RC with the QGC and the current state.

dagar, 2019-06-18:
Thanks @JMshare that's helpful.

hardeman, 2019-06-18:
> @hardeman could you describe your setup? Specific RC equipment, etc.

Taranis Q X7  +  FrSky X8R

The problem didn't occur since I mentioned it here...

@JMshare I will keep this in mind.

wluker, 2019-06-26:
I'm seeing this issue as well.
My setup ::
 PX4 version : v1.9.1
 Flight Controller : Pixhawk 2
 RC : Taranis X9D Plus + X8R
 Telemetry : 3DR

Steps to reproduce:
1. Power on the drone via battery.
2. Open QGroundControl and connect via telemetry (not usb)
3. Try to arm via RC : drone will beep with arm denial
4. In the mavlink console : issue reboot command : `reboot`
5. Repeat `reboot` and RC arming until the drone does not respond the the RC arm.

My results are sporadic, but I am seeing 4 fails in 10 tries.

wluker, 2019-06-28:
On my setup this is bug is being triggered by QGroundControl sending a `MANUAL_CONTROL` mavlink message to px4. If I disable the virtual joystick in QGroundControl this issue goes away.

julianoes, 2019-07-24:
Notes from devcall:
@davids5 agreed that we should try to reproduce that and gather more information.

@jkflying had also seen a similar report about an SBUS receiver.

Another report about this that I found.
https://discuss.px4.io/t/pixhawk-px4-ignores-rc-commands/11379

wluker, 2019-07-24:
I wonder if this is related to a comment by @dagar in https://github.com/PX4/Firmware/issues/12050#issuecomment-494119433

> only one manual input selected at boot

Is there a race condition between selecting the RC controller or the QGC virtual joystick? Sometimes it selects the RC and sometimes the virtual. Of course I don't know that others had the virtual joystick enabled in QGC like I did.

julianoes, 2019-07-31:
@hardeman @HVKHVK when you reproduce this again, can you check the output of `px4io status`? You can do that in a console or mavlink_shell.py or MAVLink shell in QGC.

julianoes, 2019-10-29:
I have to close this because there was no follow up.

ArkadiuszNiemiec, 2021-03-03:
We have the same problem. It started in 1.9, the PX4 was not responding to arming or mode changes. In 1.11 modes do change, kill switch works but still cannot arm. Only reboot fixes the problem.
@julianoes can you reopen the issue for me?
[image]
[image]

bigbellmercy, 2021-04-21:
Yes, the problem realy happened in PX4 v1.10.2. As told, after rebooting it disappeared. Thank you for the workaround.

mrpollo, 2021-04-28:
Closing issue since the original problem was not followed up upon by author @hardeman 

@HVKHVK we think your issue is unrelated, if you are still experiencing this problem please open a new issue with updated logs (latest stable, or beta) and we will happily take time to review

closed 2021-04-28 by MaEtUgR

HVKHVK, 2021-04-28:
Actually, I never got this error after I commented on this issue almost two years ago. From my point of view, the problem is because of using Pixhawk 2 and probably the cheap copies of it. I upgraded my hardware to Pixhawk 4 and I am not getting any meaningless errors I got from Pixhawk 2.