PX4/PX4-Autopilot #12255: QGC Shows Wrong Battery Percentage in v1.9.0

rollysing, 2019-06-12:
This aircraft has been flying with v1.8.0 dev which was setup in the same process as described below with no known issues at that time. I upgraded to v1.9.0 stable for some added functionalities and now have run into this issue.
Setup details:

PX4 v1.9.0 stable QGC uploaded px4_fmu-v3_default.px4
Pixhawk1 2.4.8
FX-79 Buffalo Flying Wing
3S 8000 mAh LiPo
Windows 10 Pro & QGC v3.5.2
Windows 7 Pro & QGC v3.4.4 (just to see if the same thing happens)
Telem1 connected to Teensy 3.4 for Tx telemetry
Telem2 connected to MinimOSD.
Connected Pixhawk via USB to QGC.

For a fresh setup, I installed ArduRover first to overwrite previous PX4 1.8.0dev.
Then installed PX4 v1.9.0 Stable via QGC. It uploaded px4_fmu-v3_default.px4
Calibrated all sensors, radio and setup flight modes, etc.
When setup steps are completed, unplugged USB and powered up by LiPo battery.
(Issue 1)Tx telemetry shows correct LiPo voltage but flight controller voltage only show 4.07v but actual voltage from servo rail is 5.45v.
[image]
 
(Issue 2) Plugged in USB cable but won’t connect to QGC even though the correct port shows in Device Manager. It appears but doesn’t do its normal process when the USB is plugged in, it appears for a second then disappears for a second then re-appears.
[image]

Unplugged USB and LiPo battery.
Plug in USB only and it connected to QGC then plugged in LiPo battery
(Issue 3) Battery percentage is wrong on icon bar, but if you click on the icon, the popup window shows the correct voltage.

[image]
 
(Issue 4) MinimOSD overlay is no longer showing up. Since SYS_COMPANION has been deprecated, per instructions, set MAV_1_CONFIG to TELEM 2 and Mav_1_MODE to OSD.

Originally posted in https://discuss.px4.io/t/qgc-shows-wrong-battery-percentage/11259/1 

@dagar @JulianOes @DonLakeFlyer Any guidance to resolve these issues is highly appreciated.

dagar, 2019-06-16:
@PX4/testflights have you seen anything like this?

Junkim3DR, 2019-06-19:
> @PX4/testflights have you seen anything like this?

@dagar we have not seen this issue so far.

julianoes, 2020-01-30:
@rollysing is this still an issue?

closed 2020-01-31 by rollysing