PX4/PX4-Autopilot #12258: CBRK_VELPOSERR doesn't bypass checks

tuloski, 2019-06-13:
Using this circuit breaker doesn't actually bypass the checks.
It just skips the checks, so that if the position validity was false it will never be true so the copter won't arm or takeoff or whatever it has to do with a valid position.

I don't think that is the correct behavior.
Here: https://github.com/PX4/Firmware/blob/327354705b3519472dd493e687b014ce44c6415d/src/modules/commander/Commander.cpp#L4412-L4431
I'd add at the end an:

´´´
else {
  status_flags.condition_global_position_valid = true;
  status_flags.condition_local_position_valid = true;
  status_flags.condition_local_velocity_valid = true;
}
´´´

julianoes, 2019-09-20:
Thanks for the issue. I think what you're saying is correct. Would you mind creating a pull request to fix it? Thanks.

junwoo091400, 2023-03-05:
> However, the `CBRK_VELPOSERR` param has been removed in the 1.14-beta1 branch. Perhaps we can just close this issue?

If we don't care about backporting this bugfix to the affected versions in the past, yes it would make sense to close it

junwoo091400, 2023-06-29:
Closing since we don't plan on backporting this fix for v1.13